OpenDroneMap/ODM #170: get the GPS of every pixel in orthphoto

seanXYZ, 2015-09-15:
I just use a few images from data set odm_data_seneca. The EXIF of these images contains GPS of the camera.After I got the orthphoto, we want to know the GPS of every pixel in orthphoto, what should I do? Thanks ~

dakotabenjamin, 2015-09-15:
Assuming you are using the python branch, your orthophoto tiff should be georeferenced. Open it in QGIS or another GIS program and you should be able to see the coordinates.

seanXYZ, 2015-09-16:
OK, I 'll try it~

seanXYZ, 2015-09-16:
I have git clone the newest version of the python branch.
when I install: sudo ./install.sh, it exits like this 
  > installing required packages
    - installing git submodules
Submodule path 'src/bundler': checked out 'b34cc5dcd54345272028b68f82fa4d0ae8a103c9'
Cloning into 'src/ceres-solver'...
fatal: unable to access 'https://ceres-solver.googlesource.com/ceres-solver/': Failed to connect to ceres-solver.googlesource.com port 443: Connection timed out
Clone of 'https://ceres-solver.googlesource.com/ceres-solver' into submodule path 'src/ceres-solver' 
failed

I can not access 'https://ceres-solver.googlesource.com/ceres-solver/', but I can access https://github.com/ceres-solver/ceres-solver .  Should I modify the file install.sh ?  but when I check the code, I am confused how could I do. 

echo "    - installing git submodules"
git submodule init
git submodule update

So HOW can I do to solve this problem?

seanXYZ, 2015-09-16:
I have already solve this problem:
step 1: 
     git clone  https://github.com/ceres-solver/ceres-solver 
     git clone  https://github.com/chris-cooper/bundler_sfm
    and then install them one by one
step 2: 
     copy KeyMatch to /bin or get the patch_file of bundler and build 

then we can run    run.py
finally I got the result~

closed 2015-09-16 by seanXYZ

dakotabenjamin, 2015-09-16:
calling @paulinus. It looks like when updating from earlier commits there are install problems with leftover submodules. I think you need to go into the .git/modules/src and remove ceres-solver, because now we install if from a package.

paulinus, 2015-09-16:
AFAIK `.git/modules/src` will contain leftover files for anyone having run `git submodule update` before. That should not be a problem, and git should not try to update ceres. The folder can be deleted manually, but there should be no need.  I'm not confident with submodules.

As for @seanXYZ's error, could it be that you are not at the latest commit?
The line number (384) of the error you are getting does not match the line corresponding line on the latest commit (https://github.com/OpenDroneMap/OpenDroneMap/blob/python-port/install.sh#L394).  Could you try `git pull` before `install.sh`?

seanXYZ, 2015-09-17:
OK, I 'll try later~

seanXYZ, 2015-09-17:
I run as below:
ls -a  will show me the file .gitmodules
and I modify it 
[submodule "src/bundler"]
	path = src/bundler
	url = https://github.com/chris-cooper/bundler_sfm
[submodule "src/ceres-solver"]
	path = src/ceres-solver
	url = https://ceres-solver.googlesource.com/ceres-solver

using https://github.com/ceres-solver/ceres-solver to instead  https://ceres-solver.googlesource.com/ceres-solver and all the problems gone.  Because I am in China where “google” is forbidden.（What a pity ! Maybe you think it's unbelievable, but it's really true.）
Thank you all~

paulinus, 2015-09-17:
Good that it works @seanXYZ !
You can actually delete the whole `[submodule "src/ceres-solver"]` section since the submodule is not used. Still not sure how to make that disappear from all clones.
(It is really sad that parts of the internet are blocked to you)