OpenDroneMap/ODM #171: Help about the Segmentation fault (core dumped)

xialang2012, 2015-09-15:
I installed the OpenDroneMap on my Ubuntu12.04, the installation worked well. When I used the test data download from the https://github.com/OpenDroneMap/odm_data to run the software. It showed the following errors:
  O. Tange (2011): GNU Parallel - The Command-Line Power Tool,
  ;login: The USENIX Magazine, February 2011:42-47.

This helps funding further development; and it won't cost you a cent.
Or you can get GNU Parallel without this requirement by paying 10000 EUR.

To silence this citation notice run 'parallel --bibtex' once or use '--no-notice'.

  - running bundler - Tue Sep 15 02:55:17 PDT 2015

Segmentation fault (core dumped)

quitting cause: 
	"/home/xl/opendronemap/OpenDroneMap-gh-pages/bin/bundler" "/home/xl/opendronemap/OpenDroneMap-gh-pages/reconstruction-with-image-size-2400/list.txt" --options_file "/home/xl/opendronemap/OpenDroneMap-gh-pages/reconstruction-with-image-size-2400/options.txt" > bundle/out
returned with code 35584
xl@ubuntu:~/opendronemap/

So, What should I do to resolve the problem. Thank you.
My Ubuntu owns 5GB RAM and 2 cores.

dakotabenjamin, 2015-09-15:
Try running install.sh again, then if you still have issues, try the users list to help you continue to troubleshoot:
http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users

dakotabenjamin, 2015-09-15:
Scratch that. Switch to the `python-port` branch and install that. Then use `run.py`, not `run.pl`.

smathermather, 2015-09-17:
Are we ready for existing features to merge in python-port?

xialang2012, 2015-09-17:
I download the python-port, and it showed me some errors in apt-get_install.log: E: Unable to locate package libgoogle-glog-dev.

I found that the libgoogle-glog-dev was not available on 12.04 (http://askubuntu.com/questions/312173/installing-gflags-12-04). So, May be I should to change to Ubuntu 12.10.

xialang2012, 2015-09-17:
I build the google golg on Ubuntu12.04, and run the install shell, it showed me:
> LAStools
    - installing LAStools
  < done - Thu Sep 17 06:52:43 PDT 2015

  > cmvs/pmvs
    - cleaning cmvs
    - building cmvs
    - make depend cmvs
  < done - Thu Sep 17 06:52:56 PDT 2015

  > bundler
./install.sh: line 394: cd: /home/xl/Desktop/1/OpenDroneMap-python-port/src/bundler: No such file or directory
I tried to copy the bundler.zip from master branches and unzioed it into the src directory, and run again. But it did not work. Could you give me some suggestions?

dakotabenjamin, 2015-09-17:
I'm installing 12.04 right now to look into this.

dakotabenjamin, 2015-09-17:
While I get this worked out, I would recommend updating to 14.04, which is definitely supported.

dakotabenjamin, 2015-09-17:
Progress so far: 
 * I had to manually build google-glog, then remove it from `apt-get install`
 * I had to add python-pip to `apt-get install`
 * ceres failed to build: 

> /odm_app/src/ceres-solver/internal/ceres/covariance_impl.cc:42:28: fatal error: Eigen/SparseCore: No such file or directory
compilation terminated.
make[2]: *** [internal/ceres/CMakeFiles/ceres.dir/covariance_impl.cc.o] Error 1
make[1]: *** [internal/ceres/CMakeFiles/ceres.dir/all] Error 2
make: *** [all] Error 2

One thing to note: It looks like ceres wasn't showing up in your install. Make sure you have the latest merge by calling `git pull` before installing.

xialang2012, 2015-09-19:
Thank you, now I can sucessfully built it on Ubuntu14.04. I think if we change the line 116 in install.pl to "sudo apt-get update -y > "$TOOLS_LOG_PATH/apt-get_get.log" 2>&1". So the installation shell may provide better experience for the users.

Besies, I have some troubles when I run the program. I used image of IMG_9374.JPG IMG_9375.JPG IMG_9376.JPG IMG_9377.JPG IMG_9378.JPG IMG_9379 in https://github.com/OpenDroneMap/odm_data_caliterra as the input images, and I have checked those images which can be stitched well. Then I run the run.py, it showed me as below:

>running echo "4 5" >> "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches.init.txt" && cat "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/4-5.txt" >> "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches.init.txt"
b

 >- running bundler - Sat Sep 19 06:44:55  2015
running echo "--match_table matches.init.txt
--output bundle.out
--output_all bundle_
--output_dir bundle
--variable_focal_length
--use_focal_estimate
--constrain_focal
--constrain_focal_weight 0.0
--estimate_distortion
--run_bundle" > "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/options.txt"
b
running "/home/xl/open/OpenDroneMap/bin/bundler" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/list.txt" --options_file "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/options.txt" > bundle/out
Segmentation fault (core dumped)
b

>quitting cause: 
	"/home/xl/open/OpenDroneMap/bin/bundler" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/list.txt" --options_file "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/options.txt" > bundle/out
returned with code 35584.

I do not if I should used more images to make the program run well, but the bundle module really confuses me. Thank you for your time.

smathermather, 2015-09-19:
I'm not near my computer to verify, but are we sure that Caliterra has embedded GPS? Not all datasets do.

Fi156, 2015-09-19:
Took one sample photo from the Caliterra Dataset:
http://pastebin.com/cGxyCyeG

And there was no GPS Info in the EXIF Tag.

smathermather, 2015-09-19:
Hi @xialang2012, did you use all the images in the Caliterra dataset, or a subset? Using a subset could easily result in a failure at the bundler step, as bundler will require 3 camera positions for each location in the output dataset. A subset of Caliterra has no guarantees of this completeness.

xialang2012, 2015-09-20:
@smathermather Ok, I will try it later.

dakotabenjamin, 2015-09-21:
I was able to run the caliterra dataset (both the complete dataset and the images @xialang2012 selected) without error, at least through the pmvs step. However I was running at 9 cores and 12GB RAM, so I pulled up a vagrant machine with 2 cores and 5GB RAM, and again found no errors in bundler. 

At what step of the process did you get the Extract UTM error?

xialang2012, 2015-09-22:
@dakotabenjamin  Below is the running log:
 > - configuration:
{'start_with': 'resize', 'end_with': 'odm_orthophoto', 'force_focal': None, 'odm_texturing_textureWithSize': 3600, 'matcher_threshold': 2.0, 'cmvs_maxImages': 500, 'zip_results': False, 'pmvs_threshold': 0.7, 'matcher_preselect': True, 'resize_to': 2400, 'odm_meshing_solverDivide': 9, 'use_opensfm': False, 'pmvs_wsize': 7, 'odm_texturing_textureResolution': 4096, 'matcher_ratio': 0.6, 'odm_meshing_samplesPerNode': 1, 'odm_meshing_maxVertexCount': 100000, 'pmvs_level': 1, 'run_only': None, 'odm_meshing_octreeDepth': 9, 'pmvs_csize': 2, 'matcher_kDistance': 20, 'odm_georeferencing_useGcp': False, 'odm_georeferencing_gcpFile': 'gcp_list.txt', 'matcher_useKnn': True, 'pmvs_minImageNum': 3, 'force_ccd': None}

  >- source files - Mon Sep 21 18:19:36  2015
     using IMG_9374.JPG     dimensions: 4000x3000 / focal: 4.5mm / ccd: 6.2mm
     using IMG_9375.JPG     dimensions: 4000x3000 / focal: 4.5mm / ccd: 6.2mm
     using IMG_9376.JPG     dimensions: 4000x3000 / focal: 4.5mm / ccd: 6.2mm
     using IMG_9377.JPG     dimensions: 4000x3000 / focal: 4.5mm / ccd: 6.2mm
     using IMG_9378.JPG     dimensions: 4000x3000 / focal: 4.5mm / ccd: 6.2mm
     using IMG_9379.JPG     dimensions: 4000x3000 / focal: 4.5mm / ccd: 6.2mm

    found 6 usable images

 >using max image size of 2400 x 2400

 > - preparing images - Mon Sep 21 18:19:36  2015
    resizing IMG_9374.JPG 	to /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9374.JPGrunning convert -resize 2400x2400 -quality 100 "/home/xl/open/OpenDroneMap/IMG_9374.JPG" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9374.JPG"
b
	 (2400 x 1800)
    resizing IMG_9375.JPG 	to /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9375.JPGrunning convert -resize 2400x2400 -quality 100 "/home/xl/open/OpenDroneMap/IMG_9375.JPG" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9375.JPG"
b
	 (2400 x 1800)
    resizing IMG_9376.JPG 	to /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9376.JPGrunning convert -resize 2400x2400 -quality 100 "/home/xl/open/OpenDroneMap/IMG_9376.JPG" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9376.JPG"
b
	 (2400 x 1800)
    resizing IMG_9377.JPG 	to /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9377.JPGrunning convert -resize 2400x2400 -quality 100 "/home/xl/open/OpenDroneMap/IMG_9377.JPG" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9377.JPG"
b
	 (2400 x 1800)
    resizing IMG_9378.JPG 	to /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9378.JPGrunning convert -resize 2400x2400 -quality 100 "/home/xl/open/OpenDroneMap/IMG_9378.JPG" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9378.JPG"
b
	 (2400 x 1800)
    resizing IMG_9379.JPG 	to /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9379.JPGrunning convert -resize 2400x2400 -quality 100 "/home/xl/open/OpenDroneMap/IMG_9379.JPG" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9379.JPG"
b
	 (2400 x 1800)

  >- finding keypoints - Mon Sep 21 18:19:42  2015
running "/home/xl/open/OpenDroneMap/bin/parallel" --no-notice --halt-on-error 1 -j+0 < "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/_vlsift.templist.txt"
1/6 - 524852/6 - 519584/6 - 500823/6 - 509476/6 - 412415/6 - 47311b

 > - matching keypoints - Mon Sep 21 18:21:11  2015
Error when performing utm_extraction, dumping log file:

>Error : No such file
in file '/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/IMG_9374.jpg'
Error in OdmExtractUtm:
No valid GPS position found
For more detailed information, see log file.

>running "/home/xl/open/OpenDroneMap/bin/parallel" --no-notice --halt-on-error 1 -j+0 < "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/_matches_jobs.templist.txt"
.
    5950 matches (11.5%) took  3.46s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/0-1.txt	.
    3948 matches ( 7.7%) took  3.52s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/0-2.txt	.
    2238 matches ( 4.5%) took  3.52s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/0-3.txt	...
    6649 matches (13.1%) took  3.54s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/1-2.txt	.
    4116 matches ( 8.2%) took  3.45s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/1-3.txt	.
    1932 matches ( 4.1%) took  3.33s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/1-4.txt	..
    6297 matches (12.6%) took  3.44s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/2-3.txt	.
    3226 matches ( 6.8%) took  3.43s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/2-4.txt	.
    1626 matches ( 3.9%) took  3.23s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/2-5.txt	.
    3049 matches ( 7.4%) took  3.17s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/3-5.txt	.
    5072 matches (10.7%) took  3.41s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/3-4.txt	.
    5969 matches (14.5%) took  2.63s for /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/matches/4-5.txt	b
[... cut ...]
	"/home/xl/open/OpenDroneMap/bin/bundler" "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/list.txt" --options_file "/home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400/options.txt" > bundle/out
returned with code 35584.

In odm_extracting_utm_log.txt:
>imagesPath was set to: /home/xl/open/OpenDroneMap/reconstruction-with-image-size-2400
imageListFile was set to: image_list_tmp.txt
outputCoordFile was set to: utm_coord_tmp.txt
Error in OdmExtractUtm:
No valid GPS position found

dakotabenjamin, 2015-09-22:
At first glance, it looks to me like the file capitalization issue hasn't been fixed in python-port.

dakotabenjamin, 2015-09-22:
Try this on the images:

`rename 'y/A-Z/a-z/' *`

if you run into 'file already exists' errors, use the `-f` tag.

xialang2012, 2015-09-25:
@dakotabenjamin  It can work well, thank you.

dakotabenjamin, 2015-09-25:
NP

closed 2015-09-25 by dakotabenjamin