OpenDroneMap/ODM #172: Lastools snuck in...

smathermather, 2015-09-18:
Lastools is not truly open source, so it should be excised from the python branch. I will discuss with the spotscale folks about this and replacing it with PDAL / liblas (or is it laslib...) or similar.

cc @dakotabenjamin in order to keep it on my agenda.

petrasovaa, 2015-09-21:
Technically, txt2las from LAStools is LGPL, however I agree it should be changed for [txt2las] (http://www.liblas.org/utilities/txt2las.html) from libLAS or PDAL's [translate tool] (http://www.pdal.io/apps.html#translate-command).  libLAS is in packages, PDAL not yet, but if we want to support laz, either of them must be compiled with LASzip.

smathermather, 2015-09-25:
You know, I think we even had this conversation, come to think of it... . Ya, we should probably be working with PDAL instead.

closed 2016-10-12 by dakotabenjamin