OpenDroneMap/ODM #174: error in meshing with odm_data_bellus --- Python port branch

julianandres, 2015-09-21:
I have this problem into odm_meshing

  - running meshing - Mon Sep 21 17:01:53  2015
running "/home/julian/odmPython/OpenDroneMap/bin/odm_meshing" -inputFile /home/julian/pruebasdataseptiempre11/odm_data_bellus/reconstruction-with-image-size-2400-results/option-0000.ply -outputFile /home/julian/pruebasdataseptiempre11/odm_data_bellus/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -logFile /home/julian/pruebasdataseptiempre11/odm_data_bellus/reconstruction-with-image-size-2400/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1 -solverDivide 9
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[pcl::io::savePLYFile] Input point cloud has no data!
Error in OdmMeshing:
Error when saving mesh to file:
/home/julian/pruebasdataseptiempre11/odm_data_bellus/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply

For more detailed information, see log file.
b

quitting cause: 
	"/home/julian/odmPython/OpenDroneMap/bin/odm_meshing" -inputFile /home/julian/pruebasdataseptiempre11/odm_data_bellus/reconstruction-with-image-size-2400-results/option-0000.ply -outputFile /home/julian/pruebasdataseptiempre11/odm_data_bellus/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -logFile /home/julian/pruebasdataseptiempre11/odm_data_bellus/reconstruction-with-image-size-2400/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1 -solverDivide 9
returned with code 256.

Help me please

dakotabenjamin, 2015-09-24:
Can you give more information? Specs of your machine, did installation finish correctly, what command did you run? 

If you run into more problems, you can post to the [OpenDroneMap-users mailing list](http://lists.osgeo.org/cgi-bin/mailman/listinfo/opendronemap-users).

julianandres, 2015-09-24:
Hi, the installation is correct, the command is ../OpenDroneMap/run.py in the folder from images odm_data_bellus. 

Thanks,

smathermather, 2015-09-25:
odm_data_bellus is a problematic dataset. I am hoping to name and blame the bad datasets soon in their repo. @dakotabenjamin -- do you have a recommended test dataset?

dakotabenjamin, 2015-09-25:
I only had problems with bellus before we had the georeferenced photos. I do not know if there are any problems with the dataset now.

dakotabenjamin, 2015-09-28:
Scratch that, now I am having problems too. Use the Seneca dataset, it gives me the most reliable results.

dakotabenjamin, 2015-12-09:
Alternatively, increase `--min-num-features` to 6000 or 8000 and report if that changes anything.

closed 2016-10-12 by dakotabenjamin