OpenDroneMap/ODM #177: Python Port produce only errors here.

wolkstein, 2015-09-23:
i am on ubuntu. the python port run.py produce only errors and abort. the install script ends without errors. so what do i wrong?

i try to generate something using existing foto sets that successfully ends with textured obi files using the old run.pl script.
 
```
--no-notice --halt-on-error 1 -j+0 < "/home/wolke/Bilder/opendronemap-test/Arbeit/pups/reconstruction-with-image-size-2400/_vlsift.templist.txt"
b

  - matching keypoints - Thu Sep 24 00:14:34  2015
Traceback (most recent call last):
  File "/home/wolke/develop/git/OpenDroneMap/run.py", line 976, in <module>
    func()
  File "/home/wolke/develop/git/OpenDroneMap/run.py", line 553, in match
    preselected_pairs = knnMatch_exif.preselect_pairs(BIN_PATH + "/odm_extract_utm", jobOptions["step_2_filelist"], args.matcher_kDistance, args.matcher_useKnn)
  File "/home/wolke/develop/git/OpenDroneMap/knnMatch_exif.py", line 49, in preselect_pairs
    utm_process = subprocess.Popen(utm_cmd, stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
  File "/usr/lib/python2.7/subprocess.py", line 710, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1335, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory
```

wolkstein, 2015-09-23:
sorry, my fault.
install.sh failed on opencv compilation. install.sh never finished in my python-port branch.
i can fix this problem by removing opencv build from install.sh and using my existing opencv installation in /usr/lib/python2.7/dist-packages.

now it works:)

dakotabenjamin, 2015-09-24:
Do you think OpenCV failed because you already had it installed?

wolkstein, 2015-09-24:
no, because my crappy ppa s. i use kxstudio beside an special kdenlive ppa. both mixed up my opencv dependency's:( for ffmpg h.264 codec.
but no problem at least because i can edit run.py to use my system opencv installation.
/g
wolke

smathermather, 2015-09-25:
(I assume it's ok to close this issue. Re-open @wolkstein if this is an incorrect assumption.)

closed 2015-09-25 by smathermather