OpenDroneMap/ODM #178: Problems with installation-step building odm_texturing

shomy4, 2015-10-02:
I have a problem with isntalling odm at step odm_texturing.
Im installing it on Ubuntu 14.04 8GB RAM and cpu: Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz.
I looked at odm_texturing_2_build.log and the log says:

    Scanning dependencies of target odm_texturing
    [ 25%] Building CXX object CMakeFiles/odm_texturing.dir/src/main.cpp.o
    [ 50%] Building CXX object CMakeFiles/odm_texturing.dir/src/modifiedPclFunctions.cpp.o
    [ 75%] Building CXX object CMakeFiles/odm_texturing.dir/src/Logger.cpp.o
    [100%] Building CXX object CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o
    Linking CXX executable odm_texturing
    CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::loadCameras()': 
    OdmTexturing.cpp:(.text+0x3812): undefined reference to `cv::imread(cv::String const&, int)'
    CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function   ` 
    OdmTexturing::createTextures()':
    OdmTexturing.cpp:(.text+0x782e): undefined reference to `cv::imread(cv::String const&, int)'
    OdmTexturing.cpp:(.text+0x7dd4): undefined reference to `cv::imwrite(cv::String const&,     
    cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)' 
    OdmTexturing.cpp:(.text+0x805b): undefined reference to `cv::imwrite(cv::String const&,  
    cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
    collect2: error: ld returned 1 exit status
    make[2]: *** [odm_texturing] Error 1
    make[1]: *** [CMakeFiles/odm_texturing.dir/all] Error 2
    make: *** [all] Error 2

dakotabenjamin, 2015-10-02:
It looks like an OpenCV issue. Are you installing the main branch or `python-port`?

shomy4, 2015-10-02:
I tried both branches and both installations fail.
I already have openCV 2.4.8 installed.
Could that be the problem?

shomy4, 2015-10-02:
Also in odm_texturing_1_build.log im getting bunch of warnings like:

    - The imported target "vtkChartsPythonD" references the file
       "/usr/lib/libvtkChartsPythonD.so.5.8.0"
    but this file does not exist.  Possible reasons include:
    * The file was deleted, renamed, or moved to another location.
    * An install or uninstall procedure did not complete successfully.
    * The installation package was faulty and contained
       "/usr/lib/vtk-5.8/VTKTargets.cmake"
    but not all the files it references.
    -- The imported target "vtkChartsJava" references the file
      "/usr/lib/jni/libvtkChartsJava.so.5.8.0"
    but this file does not exist.  Possible reasons include:
    * The file was deleted, renamed, or moved to another location.
    * An install or uninstall procedure did not complete successfully.
    * The installation package was faulty and contained
      "/usr/lib/vtk-5.8/VTKTargets.cmake"
    but not all the files it references.
    -- The imported target "vtk" references the file
    "/usr/bin/vtk"
    but this file does not exist.  Possible reasons include:
    * The file was deleted, renamed, or moved to another location.
    * An install or uninstall procedure did not complete successfully.
    * The installation package was faulty and contained
       "/usr/lib/vtk-5.8/VTKTargets.cmake"
    but not all the files it references.

Are those dependencies or ?

persts, 2016-05-26:
I am installing on Ubuntu14.04 and have several versions of OpenCV but the installed.sh would have found/used v2.4.8.

To get the install to complete, I had to add opencv_imgcodecs to the list of opencv modules to link with by modifying the CMakeList.txt files for odm_orthophoto, odm_georef, and odm_texturing.

dakotabenjamin, 2016-05-31:
Can you post the changed file?

persts, 2016-05-31:
diff --git a/odm_georef/CMakeLists.txt b/odm_georef/CMakeLists.txt
index 9892a37..d762aff 100644
--- a/odm_georef/CMakeLists.txt
+++ b/odm_georef/CMakeLists.txt
@@ -19,7 +19,7 @@ find_package(PCL 1.7 HINTS "${PCL_DIR}/share/pcl-1.7")
 find_package(OpenCV HINTS "${OPENCV_DIR}" REQUIRED)
 
 # Only link with required opencv modules.
-set(OpenCV_LIBS opencv_core opencv_imgproc opencv_highgui)
+set(OpenCV_LIBS opencv_core opencv_imgproc opencv_highgui opencv_imgcodecs)
 
 # Add the PCL and Eigen include dirs. 
 # Necessary since the PCL_INCLUDE_DIR variable set bu find_package is broken.)
diff --git a/odm_orthophoto/CMakeLists.txt b/odm_orthophoto/CMakeLists.txt
index a858d73..efa443a 100644
--- a/odm_orthophoto/CMakeLists.txt
+++ b/odm_orthophoto/CMakeLists.txt
@@ -15,7 +15,7 @@ find_package(PCL 1.7 HINTS "${PCL_DIR}/share/pcl-1.7" REQUIRED)
 find_package(OpenCV HINTS "${OPENCV_DIR}" REQUIRED)
 
 # Only link with required opencv modules.
-set(OpenCV_LIBS opencv_core opencv_imgproc opencv_highgui)
+set(OpenCV_LIBS opencv_core opencv_imgproc opencv_highgui opencv_imgcodecs)
 
 # Add the PCL, Eigen and OpenCV include dirs. 
 # Necessary since the PCL_INCLUDE_DIR variable set by find_package is broken.)
diff --git a/odm_texturing/CMakeLists.txt b/odm_texturing/CMakeLists.txt
index 9838db0..837dc48 100644
--- a/odm_texturing/CMakeLists.txt
+++ b/odm_texturing/CMakeLists.txt
@@ -17,7 +17,7 @@ find_package(PCL 1.7 HINTS "${PCL_DIR}/share/pcl-1.7" REQUIRED)
 find_package(OpenCV HINTS "${OPENCV_DIR}" REQUIRED)
 
 # Only link with required opencv modules.
-set(OpenCV_LIBS opencv_core opencv_imgproc opencv_highgui)
+set(OpenCV_LIBS opencv_core opencv_imgproc opencv_highgui opencv_imgcodecs)
 
 # Add the PCL, Eigen and OpenCV include dirs. 
 # Necessary since the PCL_INCLUDE_DIR variable set by find_package is broken.)

closed 2016-10-12 by dakotabenjamin