OpenDroneMap/ODM #181: Python-port: odm_georeferencing, odm_orthophoto

rionlerm, 2015-10-03:
Please assist...

...run.py with --use-opensfm True and --odm_georeferencingGcp False produces no orthophoto. The arguments do produce .ply meshes and texturing with mosaicked photos on top though?
odm_georeferencing folder under results also then contain no files.
Also ran ./install.sh recently through python-port.

Terminal output shows that I must specify ...Gcp False even though I did include the argument.

Run from Uhuntu 14

#python

shomy4, 2015-10-07:
I have the same problem.
What parameters should i use to get GeoTiff image, if that is even possible?

rionlerm, 2015-10-07:
@dakotabenjamin mentioned there might be a bug, but I hope it gets solved soon.

rionlerm, 2015-10-14:
@smathermather could you assist with above?

smathermather, 2015-10-15:
@rionlerm -- I have been traveling a lot. I have a backlog but hope to dive into the python port soon.

rionlerm, 2015-10-15:
Thanks, seemed to me the issues went quiet. Thought it was something I said
;)

Regards,
*Rion Lerm (jnr.)*
(013) 735 3536
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)
Click here to learn about Rion's responsibilities within SAEON
<http://www.saeon.ac.za/enewsletter/archives/2014/october2014/doc09>
Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2015-10-27:
Possible problems I've come up with: 

1. GPS EXIF is missing or corrupted in one or more of the photos
2. GPS is in the wrong units or projection

dakotabenjamin, 2015-10-27:
The following python code will tell you if any of the photos in a directory are missing GPS EXIF data: 
```
import exifread
import glob

images = glob.glob("./*.JPG")

for image in images:
    f = open(image, 'rb')
    tags = exifread.process_file(f, details=False)
    if 'GPS GPSLatitude' not in tags.keys():
        print '%s does not have GPS EXIF' % image
````

dakotabenjamin, 2015-10-27:
Also, -odm_georeferencing-useGcp False is unnecessary because it is False by default. Have you tried without using this parameter?

rionlerm, 2015-10-27:
Hi and thanks for looking into the problem. I will re-run without
$-odm_georeferencing-useGcp
False. Can I run the python code through Ubuntu's terminal as when running
run.py? to check EXIF etc.

Regards,
*Rion Lerm (jnr.)*
(013) 735 3536
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)
Click here to learn about Rion's responsibilities within SAEON
<http://www.saeon.ac.za/enewsletter/archives/2014/october2014/doc09>
Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2015-10-27:
your best bet is to `cd` into your images directory, the go into the python command line by typing `python`, then pasting the code above.

rionlerm, 2015-10-27:
OK, python code but, I am doing something wrong. No output:
[image: Inline images 1]

...running run.py *without* -odm_georeferencing-useGcp False. Also find
attached only log file produced:
[image: Inline images 2]

Regards,
*Rion Lerm (jnr.)*
(013) 735 3536
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)
Click here to learn about Rion's responsibilities within SAEON
<http://www.saeon.ac.za/enewsletter/archives/2014/october2014/doc09>
Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

imagesPath was set to: /home/rion2/Desktop/SAEON/
Error in OdmExtractUtm:
Argument '-imageListFile' has a bad value (file not accessible).

dakotabenjamin, 2015-10-27:
Images didn't attach from email. Could you edit your comment on GitHub?

rionlerm, 2015-10-27:
Also:
[image: Inline images 1]

Regards,
*Rion Lerm (jnr.)*
(013) 735 3536
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)
Click here to learn about Rion's responsibilities within SAEON
<http://www.saeon.ac.za/enewsletter/archives/2014/october2014/doc09>
Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

rionlerm, 2015-10-27:
...running run.py without -odm_georeferencing-useGcp False. Also find attached only log file produced:

  using max image size of 2400 x 2400

  - running georeferencing - Tue Oct 27 17:52:04  2015
running "/media/rion2/ce98ee14-8b56-461c-a977-1ff57c75261e/rion/OpenDroneMap/bin/odm_extract_utm" -imagesPath /home/rion2/Desktop/SAEON/ -imageListFile /home/rion2/Desktop/SAEON/reconstruction-with-image-size-2400/pmvs/list.rd.txt -outputCoordFile /home/rion2/Desktop/SAEON/reconstruction-with-image-size-2400/odm_georeferencing/coordFile.txt
Error in OdmExtractUtm:
Argument '-imageListFile' has a bad value (file not accessible).
For more detailed information, see log file.
b

quitting cause: 
	"/media/rion2/ce98ee14-8b56-461c-a977-1ff57c75261e/rion/OpenDroneMap/bin/odm_extract_utm" -imagesPath /home/rion2/Desktop/SAEON/ -imageListFile /home/rion2/Desktop/SAEON/reconstruction-with-image-size-2400/pmvs/list.rd.txt -outputCoordFile /home/rion2/Desktop/SAEON/reconstruction-with-image-size-2400/odm_georeferencing/coordFile.txt
returned with code 256.

rion2@rion2-ThinkPad-Edge-E420s:~/Desktop/SAEON$

Secondly doing something wrong with python: 

rion2@rion2-ThinkPad-Edge-E420s:~/Desktop/SAEON$ python
Python 2.7.6 (default, Jun 22 2015, 17:58:13) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import exifread
>>> import glob
>>> 
>>> images = glob.glob("./*.JPG")
>>> 
>>> for image in images:
...     f = open(image, 'rb')
...     tags = exifread.process_file(f, details=False)
...     if 'GPS GPSLatitude' not in tags.keys():
...         print '%s does not have GPS EXIF' % image
... 
>>>

rionlerm, 2015-10-27:
Last e-mail was screenshot showing GPS data of one image as pasted into
e-mail body.

I copied code to GitHub

Regards,
*Rion Lerm (jnr.)*
(013) 735 3536
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)
Click here to learn about Rion's responsibilities within SAEON
<http://www.saeon.ac.za/enewsletter/archives/2014/october2014/doc09>
Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2015-10-27:
They python looks fine: there will be no output if all the images have GPS Latitude.

dakotabenjamin, 2015-10-27:
@merkato was having a similar problem. Did you ever solve it?

merkato, 2015-10-28:
No, I never solved problems with Boruszyn dataset orthophoto generation. Right now I'm installing my Ubuntu after not so happy apt-get upgrade ;) After my workstation will be ready, I will try to reproduce this, and ask for debug help.

dakotabenjamin, 2015-10-28:
@rionlerm could you post your install log for odm_extract_utm? It's under your odm installation directory logs/odm_extract_2_build.log

rionlerm, 2015-10-29:
odm_extract_utm_2_build.log:

Scanning dependencies of target odm_extract_utm
[ 33%] Building CXX object CMakeFiles/odm_extract_utm.dir/src/main.cpp.o
[ 66%] Building CXX object CMakeFiles/odm_extract_utm.dir/src/Logger.cpp.o
[100%] Building CXX object CMakeFiles/odm_extract_utm.dir/src/UtmExtractor.cpp.o
Linking CXX executable odm_extract_utm
[100%] Built target odm_extract_utm

merkato, 2015-10-29:
My results right now. 

 - running georeferencing - Thu Oct 29 08:52:23  2015
running "/home/mechanik/dev/OpenDroneMap/bin/odm_georef" -bundleFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/pmvs/bundle.rd.out -gcpFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/gcp_list.txt -imagesPath /home/mechanik/dev/odm_data/odm_boruszyn_kap/ -imagesListPath /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/pmvs/list.rd.txt -bundleResizedTo 2400 -inputFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj -outputFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj -outputCoordFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/coordFile.txt -inputPointCloudFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000.ply -outputPointCloudFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000_georef.ply -logFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo_georef_system.txt
b

  - running orthophoto generation - Thu Oct 29 08:52:48  2015
running "/home/mechanik/dev/OpenDroneMap/bin/odm_orthophoto" -inputFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj -logFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_orthophoto/odm_orthophoto_log.txt -outputFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_orthphoto.png -resolution 20.0 -outputCornerFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_orthphoto_corners.txt
b
    Warning: No geo-referenced orthophoto created due to missing geo-referencing or corner coordinates.

  - done - Thu Oct 29 08:53:50  2015

------------------------------
/home/mechanik/dev/OpenDroneMap/bin/odm_georef_log.txt
Arguments given
-bundleFile
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/pmvs/bundle.rd.out
-gcpFile
/home/mechanik/dev/odm_data/odm_boruszyn_kap/gcp_list.txt
-imagesPath
/home/mechanik/dev/odm_data/odm_boruszyn_kap/
-imagesListPath
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/pmvs/list.rd.txt
-bundleResizedTo
2400
-inputFile
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj
-outputFile
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj
-outputCoordFile
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/coordFile.txt
-inputPointCloudFile
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000.ply
-outputPointCloudFile
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000_georef.ply
-logFile
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt
-georefFileOutputPath
/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo_georef_system.txt

Reading cameras from: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/pmvs/bundle.rd.out
Reading GCPs from: /home/mechanik/dev/odm_data/odm_boruszyn_kap/gcp_list.txt
Images location is set to: /home/mechanik/dev/odm_data/odm_boruszyn_kap/
Reading image list from: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/pmvs/list.rd.txt
Bundle resize value is set to: 2400
Reading textured mesh from: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj
Writing output to: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj
Exporting cameras georeferenced gps positions to: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/coordFile.txt
Reading point cloud from: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000.ply
Writing output to: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000_georef.ply
Log file path was set to: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt
Georef file output path is set to: /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo_georef_system.txt

Reading following GCPs from file:
x_: 340577 y_: 547925 z_: 70 pixelX_: 3498 pixelY_: 691 image: img_4851.jpg
x_: 340610 y_: 548131 z_: 71 pixelX_: 2921 pixelY_: 1248 image: img_4854.jpg
x_: 340649 y_: 548176 z_: 70 pixelX_: 2312 pixelY_: 1078 image: img_4912.jpg

[... cut ...]
Saving point cloud file to '/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000_georef.ply'...
.. point cloud file saved.

Saving georeferenced camera positions to /home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/coordFile.txt
...coordinate file saved.

Saving georeference system file to '/home/mechanik/dev/odm_data/odm_boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo_georef_system.txt'...
... georeference system saved.
-----------------------------------

rionlerm, 2015-10-29:
So orthophoto issues with and without gcp_file. I did not use the latter: bargaining on EXIF info on my GPS-tagged photos

dakotabenjamin, 2015-10-29:
@merkato I have the same problem with the boruszyn_kap dataset. It looks like its not reading the odm_orthphoto_corners.txt file

@rionlerm yes it seems the issue is slightly different, but in the same area. I've pinged the devs of the georeferencing function for some additional help.

dakotabenjamin, 2015-10-29:
@rionlerm can you post this file: `/home/rion2/Desktop/SAEON/reconstruction-with-image-size-2400/pmvs/list.rd.txt`

rionlerm, 2015-10-30:
list.rd.txt:

./dji_0007.jpg
./dji_0008.jpg
./dji_0009.jpg
./dji_0010.jpg
./dji_0011.jpg
./dji_0012.jpg
./dji_0013.jpg
./dji_0014.jpg
./dji_0015.jpg
./dji_0016.jpg
./dji_0017.jpg
./dji_0018.jpg
./dji_0019.jpg
./dji_0020.jpg
./dji_0021.jpg
./dji_0022.jpg
./dji_0023.jpg
./dji_0024.jpg
./dji_0025.jpg
./dji_0026.jpg

results:

total 2788
drwxrwxr-x 5 rion2 rion2    4096 Oct 30 11:57 .
drwxrwxr-x 7 rion2 rion2    4096 Oct 30 12:03 ..
-rw-rw-r-- 1 rion2 rion2 2797425 Oct 30 11:57 bundle.rd.out
-rw-rw-r-- 1 rion2 rion2     394 Oct 30 11:57 centers-0000.ply
-rw-rw-r-- 1 rion2 rion2     858 Oct 30 11:57 centers-all.ply
-rw-rw-r-- 1 rion2 rion2     300 Oct 30 11:57 list.rd.txt
drwxrwxr-x 2 rion2 rion2    4096 Oct 30 12:03 models
-rw-rw-r-- 1 rion2 rion2     224 Oct 30 11:57 option-0000
-rw-rw-r-- 1 rion2 rion2     131 Oct 30 11:57 pmvs_options.txt
-rw-rw-r-- 1 rion2 rion2      25 Oct 30 11:57 pmvs.sh
-rw-rw-r-- 1 rion2 rion2    2676 Oct 30 11:57 prep_pmvs.sh
-rw-rw-r-- 1 rion2 rion2      47 Oct 30 11:57 ske.dat
drwxrwxr-x 2 rion2 rion2    4096 Oct 30 11:57 txt
-rw-rw-r-- 1 rion2 rion2     358 Oct 30 11:57 vis.dat
drwxrwxr-x 2 rion2 rion2    4096 Oct 30 11:57 visualize

dakotabenjamin, 2015-10-30:
try the following: 
```
chmod 777 /home/rion2/Desktop/SAEON/reconstruction-with-image-size-2400/pmvs/list.rd.txt
run.py <whatever you used before> --start-with odm_georeferencing
```

rionlerm, 2015-10-30:
Hi, I ran chmod 777 for list.rd.txt and then as usual by reverting back to directory containing images (last line shows code I run:

found 20 usable images

  using max image size of 2400 x 2400

  - running georeferencing - Fri Oct 30 15:08:43  2015
Warning: No GCP file. Consider rerunning with argument --odm_georeferencing-useGcp false --start-with odm_georeferencing
Skipping orthophoto

  - done - Fri Oct 30 15:08:43  2015
rion2@rion2-ThinkPad-Edge-E420s:~/Desktop/SAEON$ ../../../../media/rion2/ce98ee14-8b56-461c-a977-1ff57c75261e/rion/OpenDroneMap/run.py --use-opensfm True --odm_georeferencing-useGcp False --start-with odm_georeferencing

Result same as before.

dakotabenjamin, 2015-10-30:
Don't use `--odm_georeferencing-useGcp False` that parameter is set to `False` by default, and for some reason I think calling it as a parameter flags it as `True`

rionlerm, 2015-10-30:
It worked! You are genius. I just want to check the georeferencing of GeoTIFF in GIS.

Well done.

dakotabenjamin, 2015-10-30:
Wow ok. So this does bring up an interesting problem about permissions. @smathermather any ideas?

smathermather, 2015-11-01:
(BTW, I rarely recommend ```chmod 777``` but either ```chmod 700``` or ```chmod 770```. Just leave that world flag, the third number as zero always).

Anyway, if the user running the script can't read ```list.rd.txt``` then naturally things will fail. I'm really unclear why the permissions on this would be set wrong. @rionlerm -- is this perhaps running on a VM and the shared drive on the VM is NTFS? Then their could be issues with the files inheriting weird permissions upon creation maybe... . Just a shot in the dark here.

dakotabenjamin, 2015-11-02:
@smathermather interesting because all of the files created by `run.py` have `777` permissions.

smathermather, 2015-11-03:
Heheh. That needs fixed... . That's a pet peeve of mine... .

rionlerm, 2015-11-09:
@smathermather I am not using a VM. Shall I attempt with chmod 700 or has there been other advances made with regards to this issue?

smathermather, 2015-11-10:
@rionlerm -- I'm not clear on where the permission are getting set, but ```chmod 700 ...``` or 770 would be fine for now.

rionlerm, 2015-11-11:
Hi @smathermather  and @dakotabenjamin . I recently ran the script using 75 images taken with the DJI Phantom 3 Advanced (4000x3000 pixels/photo). I used the following script first:

run.py --use-opensfm True --run-only odm_orthophoto

It quitted the cause due to seeming memory issues: it ran out of memory.

I then ran: chmod 700 pmvs/list.rd.txt which executed fine.

Lastly, I ran the the first without ...--run-only, just --use-opensfm True and output was:

    Creating geo-referenced LAS file (expecting warning)...
running "/media/rion2/ce98ee14-8b56-461c-a977-1ff57c75261e/rion/OpenDroneMap/bin/txt2las" -i /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400-results/option-0000_georef.ply -o /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400-results/pointcloud_georef.laz -skip 30 -parse xyzRGBssss -set_scale 0.01 0.01 0.01 -set_offset 298319 7381106 0 -translate_xyz 298319 7381106 0 -epsg 32736
WARNING: written 2219493 points but expected 0 points
b

  - running orthophoto generation - Tue Nov 10 16:41:05  2015
running "/media/rion2/ce98ee14-8b56-461c-a977-1ff57c75261e/rion/OpenDroneMap/bin/odm_orthophoto" -inputFile /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj -logFile /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400/odm_orthophoto/odm_orthophoto_log.txt -outputFile /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400-results/odm_orthphoto.png -resolution 20.0 -outputCornerFile /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400/odm_orthphoto_corners.txt
OpenCV Error: Insufficient memory (Failed to allocate 6254270596 bytes) in OutOfMemoryError, file /media/rion2/ce98ee14-8b56-461c-a977-1ff57c75261e/rion/OpenDroneMap/src/opencv/modules/core/src/alloc.cpp, line 52
Error in OdmOrthoPhoto:
/media/rion2/ce98ee14-8b56-461c-a977-1ff57c75261e/rion/OpenDroneMap/src/opencv/modules/core/src/alloc.cpp:52: error: (-4) Failed to allocate 6254270596 bytes in function OutOfMemoryError

b

quitting cause: 
	"/media/rion2/ce98ee14-8b56-461c-a977-1ff57c75261e/rion/OpenDroneMap/bin/odm_orthophoto" -inputFile /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj -logFile /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400/odm_orthophoto/odm_orthophoto_log.txt -outputFile /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400-results/odm_orthphoto.png -resolution 20.0 -outputCornerFile /home/rion2/Desktop/SAEON/letaba_project_river_2015/reconstruction-with-image-size-2400/odm_orthphoto_corners.txt
returned with code 256.

dakotabenjamin, 2015-11-12:
How many GB of RAM do you have on your machine?

rionlerm, 2015-11-16:
Ooh, only 4GB, i7 processor and Ubuntu 14

Regards,
*Rion Lerm (jnr.)*
(013) 735 3536
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)
Click here to learn about Rion's responsibilities within SAEON
<http://www.saeon.ac.za/enewsletter/archives/2014/october2014/doc09>
Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2015-11-16:
It's trying to allocate 6 GB, which is why it failed. We are still trying to get OpenCV to work properly, so (hopefully) in the next update this will be resolved.

rionlerm, 2015-11-16:
Cool, thanks. How can I get updates on next update?

Regards,
*Rion Lerm (jnr.)*
(013) 735 3536
Ndlovu Node
Phalaborwa Gate, Kruger National Park
South African Environmental Observation Network (SAEON)
Click here to learn about Rion's responsibilities within SAEON
<http://www.saeon.ac.za/enewsletter/archives/2014/october2014/doc09>
Click here to learn about SAEON <http://www.saeon.ac.za/saeon-background>

dakotabenjamin, 2015-11-16:
I will post to the mailing lists and update the README when it happens.

smathermather, 2015-11-23:
I think we can close this with https://github.com/OpenDroneMap/OpenDroneMap/pull/203 . Will reopen if necessary.

closed 2015-11-23 by smathermather