OpenDroneMap/ODM #182: enquire suggestions about the mosaic of NDVI

xialang2012, 2015-10-10:
Hi everyone,

I used the OpenDroneMap to process the true color image, and it worked well when the quality of images is good. Recently I bought a near-infrared camera which owns green band, red band, blue band and near-infrared band. What I hope to obtain is the mosaic (orthophoto) NDVI images which is calculated by the following equation (1)
NDVI = (Nir-R)/(Nir+R)
in the equation, Nir is the reflectance of the near-infrared band, R is the reflectance of the Red band. Currently, I have one ideas to hand the mosaic work. The first step is calculate the NDVI by using the equation (1), and then use the NDVI images as the input images. However, the equation (1) is not a linear transformation, and especially when the vegetation grows well, the NDVI images usually can not be matched well by the current image recognition algorithms, such as sift, harris etc. 

So, I wonder if I can process the RGB images first and record the relationships between the images, and then calculate NDVI for each images, and use those relationships to mosaic the NDVI images?

Thanks all of you.

smathermather, 2015-10-10:
As you surmised, you would mosaic first and then calculate NDVI, otherwise feature extraction could be problematic.

xialang2012, 2015-10-10:
@smathermather Yeah, but the mosaic results between the Red image and Nir images may be different, for the image features for those two band is different.

smathermather, 2015-10-14:
@xialang2012 -- that's true for any two bands. Is it especially so between NIR and the other bands?

xialang2012, 2015-10-15:
@smathermather Yes, there are many difference between NIR and Red bands. I wonder how the other commercial softwares, e.g. Pix4d achieve these functions.

smathermather, 2015-10-15:
My hunch is that the differences between the bands don't matter much, insofar as finding any static features to match in any bands or combination of bands is all that is needed.

If there is substantial differences in bidirectional reflectance distribution function ( https://en.wikipedia.org/wiki/Bidirectional_reflectance_distribution_function#Related_functions ) or other anisotropy between bands sufficient to result in non-static features where otherwise features could be matched, then it's an issue. But, I suspect for most use cases, the differences can be ignored.

That said, I'd love to see counter examples, especially in a CIR video feed which demonstrate between band anisotropy which shifts or changes features shape such that matches are impossible. My mental model could be quite wrong!

xialang2012, 2015-10-15:
@smathermather  I will used the Red and Nir band images to try it later. By the way, I found the SURF feature match algorithm is fast than SIFT in the OpenCV library. Can we replace SIFT with SURF in the  OpenDroneMap?

dakotabenjamin, 2015-10-16:
I've read that SURF, while faster, detects fewer features and has fewer matches. I'm not sure we want to trade speed for lower effectiveness.

xialang2012, 2015-10-18:
It is true that SURF detects fewer features that SIFT. So far, the most efficient mosaic software I have used is ICE from Microsoft. But Microsoft did not show enough detail information about the algorithm used in the ICE.

smathermather, 2015-11-23:
@xialang2012 -- let me know if this issue needs reopened.

closed 2015-11-23 by smathermather

Sharathmk99, 2016-07-16:
@xialang2012 I'm also trying to get NDVI images from OpenDroneMap. Please help if you have successfully complete. Please share

xialang2012, 2016-07-18:
@Sharathmk99 Sorry, not yet.

smathermather, 2016-07-18:
Hi @Sharathmk99  -- can you explain more what you are trying to do?

hblanken, 2017-01-24:
@xialang2012, @smathermather - Just checking in again to see how we can succeed with creating NDVI maps with ODM? Is there any plugin or code completed by now that you can point me to?

LucasMM86, 2017-01-24:
This is related to #190  , did you see that?