OpenDroneMap/ODM #184: Does OpenDroneMap support semi global matching

JzHuai0108, 2015-10-15:
Hello,

It seems to me OpenDroneMap only supports patched based methods like CMVS and PMVS to densify point clouds. Following generating point cloud with PMVS, semi global matching (SGM) was reported to improve the density of point cloud. It seems both Agisoft PhotoScan and Pix4D support SGM.

Is OpenDroneMap capable of performing SGM after PMVS? If not, is there a plan to incorporate SGM into OpenDroneMap?

Thanks,

dakotabenjamin, 2015-10-15:
Hi,

I have been researching SGM and looking at ways to incorporate it into our code. Right now, however, there are no concrete plans.

smathermather, 2015-10-18:
Good tip, @JzHuai0108. This is definitely something to look into given how dense Photoscan and Pix4D clouds are relative to ODM.

smathermather, 2015-11-23:
SGM is definitely on the wish list. It's lack of implementation here is probably the greatest gap in the current toolchain.

edgarriba, 2015-12-04:
maybe you can take a look here
https://github.com/Itseez/opencv/blob/master/modules/calib3d/src/stereosgbm.cpp

smathermather, 2015-12-06:
Yes, OpenCV has it. Here's some notes from @paulinus I got via e-mail:

"The original paper

  http://www.dlr.de/rm/Portaldata/52/Resources/dokumente/modeler/cvpr05hh.pdf

or it journal version

  http://elib.dlr.de/55367/1/Stereo_Processing-Hirschm%C3%BCller.pdf

are still the reference for this as far as i know.

There is an implementation of the method in opencv, but only for the case of 2 images.

 http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html#stereosgbm

For drone images, we would need to handle more images and to split each image into tiles since SGM eats a lot of memory for high resolution images."

dakotabenjamin, 2016-10-12:
#389 supports SGM

closed 2016-10-12 by dakotabenjamin