OpenDroneMap/ODM #185: Can OpenDroneMap process images collected by different cameras? Does prior trajectory come in handy?

JzHuai0108, 2015-10-15:
I tried opendronemap with up to a hundred of UAV images, without ground control points, it achieved accuracy close to commercial software like Agisoft PhotoScan. So I want to explore more about OpenDroneMap.

(1) Is It possible for OpenDroneMap to process images collected by cameras of different specs at once? 
Of course, I can generate point clouds from different image datasets separately, geotag each of them, and put them together. But this requires the knowledge of geolocation and is time consuming and possibly less accurate.

(2) By the way, if we have a priori trajectory of the camera, will this prior information benefit generating point clouds? E.g., reducing computation time, improving convergence accuracy, or even eliminating the need of bundle adjustment.

Thanks for your patience!

smathermather, 2015-10-18:
Hi @JzHuai0108,

(1) I believe so, although I haven't tested myself (yet)
(2) Computation time on the (still somewhat broken but soon to be fixed) python branch is significantly reduced with GPS data. I don't know that it helps much with convergence accuracy, and there is no assumption that your camera ephemeris is good enough to avoid bundle adjustment.

I hope this helps!

smathermather, 2015-11-23:
Hi @JzHuai0108 -- current primary branch should address both questions. Cheers.

closed 2015-11-23 by smathermather

dakotabenjamin, 2015-11-23:
it should be noted that knn matching (what we use) is technically different from using trajectories to improve bundling.

smathermather, 2015-11-24:
Ah, good point @dakotabenjamin. Trajectories might be addressed by some SLAM work that's in the pipeline, if I understand the question.

JzHuai0108, 2015-11-24:
I will try the latest version of OpenDroneMap with regard to given poses of cameras. These known camera poses may be inaccurate, for example, they may be computed by a coarse GPS/INS module, or a visual odometry method like ORB_SLAM. If I get time, I will do a comparison between (1) OpenDroneMap on images without camera poses, and (2) OpenDroneMap on images with GPS coordinates and rough estimates of camera attitude.