OpenDroneMap/ODM #186: opensfm bundle.out creates only a subset sparse cloud

dakotabenjamin, 2015-10-20:
When running using OpenSfM, the following problem occurs: 

In the opensfm folder, multiple bundler files are created: 
```
bundle_r000.out
bundle_r001.out
bundle_r002.out
...
```
These out files each contain separate sparse point clouds, but [OpenSfM](https://github.com/OpenDroneMap/OpenDroneMap/blob/python-port/run.py#L705) only passes bundle_r000.out to pmvs. @paulinus, how do we ensure all of the data is passed to the next step? By concatenating the .out files? some other method? Otherwise we will be unable to process large datasets.

paulinus, 2015-10-20:
OpenSfM tries to build a single reconstruction, but sometimes this is not possible.

A reconstruction is build incrementally by adding one image at a time.  It can happen that some images can not be added to a reconstruction due to little overlap with the images in the reconstruction. When this happens, OpenSfM starts a new, independent reconstruction with the remaining images. So the final result is a set of independent reconstructions.

Bundler does keep only the first reconstruction. So when exporting to bundler, we have to choose one of the reconstructions produced by OpenSfM. Currently the largest one is chosen (they are sorted by size), and the other are discarded.

I think that is an acceptable behavior. An alternative would be to run the rest of the pipeline for each of the reconstructions, but the result would be a set of independent meshes and we would need to modify the pipeline a bit.

dakotabenjamin, 2015-10-20:
That is interesting. Is there an option in OpenSfM to reduce the threshold for overlapping images? I'm getting reconstructions that are 1/4 the possible size when run using other reconstruction software, even when there is ~90% overlap.

paulinus, 2015-10-21:
The is no such option. If you are getting these results with large overlaps, it is probably a bug. Are the images available? I would like to check what is going on.

dakotabenjamin, 2015-10-21:
I will post them to the odm_data repo today. I've also noticed the issue occurs on [odm_data_rvnir](https://github.com/OpenDroneMap/odm_data_rv_nir), which had at least 85% overlap/sidelap.

dakotabenjamin, 2015-10-21:
For reference, here's what I got from odm:

[image]

And here's a dense point cloud from Agisoft Photoscan:

[image]

smathermather, 2015-10-21:
@dakotabenjamin -- "closed source" software... :)

paulinus, 2015-10-21:
@dakotabenjamin, I've tried the `odm_data_rv_nir` dataset. With the default values there are too few keypoints matched between overlapping images.

I suspect this is because vegetation is tricky. There are a lot of small visual features and their appearance changes a lot between different viewpoints.

I've been able to get a reconstruction with 549 images by reducing the image size and increasing the number of features.  With this `config.yaml` file
```
feature_process_size: 1024
feature_min_frames: 8000
matching_gps_neighbors: 10
processes: 4
```
I'm getting
[image]

There is some global bending, but othewise looks ok. I'll try the other dataset and let you know how it goes. Then will think on either changing the default parameterss or make them availablen from `run.py`

paulinus, 2015-10-21:
Got 245 images reconstructed for `odm_data_waterbury` with the config above.

[image]

I need to update the defaults in `run.py`

dakotabenjamin, 2015-11-03:
How is this coming? Once this change is pushed, I think I am going to merge `python-port` into the master branch.

paulinus, 2015-11-03:
I have a branch updating opensfm here https://github.com/paulinus/OpenDroneMap/tree/update-opensfm will finish some details and pull-request today.

smathermather, 2015-11-23:
This should be closed with https://github.com/OpenDroneMap/OpenDroneMap/pull/203 , yes?

closed 2015-11-23 by smathermather