OpenDroneMap/ODM #188: install.sh doesn't finish properly

edgarriba, 2015-10-27:
Hi,

I'm trying to install the project using the installation file but it stops at texturing without errors.
Any idea? Thanks in advance !

Screen log:

  > pcl 
    - configuring pcl
    - building and installing pcl
  < done - mar oct 27 14:37:20 CET 2015

  > meshing 
    - configuring odm_meshing
    - building odm_meshing
  < done - mar oct 27 14:37:40 CET 2015

  > texturing 
    - configuring odm_texturing
    - building odm_texturing

dakotabenjamin, 2015-10-27:
Please post your odm_texturing log files found in the logs directory. Thanks!

edgarriba, 2015-10-27:
Scanning dependencies of target odm_texturing
[ 25%] Building CXX object CMakeFiles/odm_texturing.dir/src/Logger.cpp.o
[ 50%] Building CXX object CMakeFiles/odm_texturing.dir/src/main.cpp.o
[ 75%] Building CXX object CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o
[100%] Building CXX object CMakeFiles/odm_texturing.dir/src/modifiedPclFunctions.cpp.o
Linking CXX executable odm_texturing
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::loadCameras()':
OdmTexturing.cpp:(.text+0x3812): undefined reference to `cv::imread(cv::String const&, int)'
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::createTextures()':
OdmTexturing.cpp:(.text+0x782e): undefined reference to `cv::imread(cv::String const&, int)'
OdmTexturing.cpp:(.text+0x7dd4): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
OdmTexturing.cpp:(.text+0x805b): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [odm_texturing] Error 1
make[1]: *** [CMakeFiles/odm_texturing.dir/all] Error 2
make: *** [all] Error 2

edgarriba, 2015-10-27:
I would add that have OpenCV 3.0 installed

dakotabenjamin, 2015-10-27:
Which branch are you using? Try switching to `python-port`

edgarriba, 2015-10-27:
main branch. Ok, trying.

edgarriba, 2015-10-28:
@dakotabenjamin got the same the error

dakotabenjamin, 2015-10-28:
OK, thanks. We need to get OpenCV contained so that other installations do not conflict. This has been a problem for other users as well.

dakotabenjamin, 2015-10-29:
@edgarriba I think #192 may fix your problem. Please test and report back.

edgarriba, 2015-10-30:
@dakotabenjamin 

    - script finished - Fri Oct 30 11:07:13 UTC 2015

dakotabenjamin, 2015-10-30:
Yay!

closed 2015-10-30 by dakotabenjamin