OpenDroneMap/ODM #189: [python-port] missing python modules

merkato, 2015-10-28:
I'm installed fresh 15.10 xubuntu, and cloned python-port branch. After installation started to work with ODM:

run.py --use-opensfm True --odm_georeferencing-useGcp True

And there's surprise, missing python modules  **xmltodict** and **csfm**

'''- running OpenSfM - Wed Oct 28 15:54:46  2015
running import_bundler
running PYTHONPATH=/home/mechanik/dev/OpenDroneMap/lib/python2.7/dist-packages "/home/mechanik/dev/OpenDroneMap/src/OpenSfM/bin/import_bundler" opensfm --list list.txt
No bundle.out given.  Importing only the list of images list.txt.
Converting output from Bundler to OpenSfM
b
running PYTHONPATH=/home/mechanik/dev/OpenDroneMap/lib/python2.7/dist-packages "/home/mechanik/dev/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
Traceback (most recent call last):
  File "/home/mechanik/dev/OpenDroneMap/src/OpenSfM/bin/focal_from_exif", line 13, in <module>
    from opensfm import exif
  File "/home/mechanik/dev/OpenDroneMap/src/OpenSfM/opensfm/exif.py", line 8, in <module>
    import xmltodict as x2d
ImportError: No module named xmltodict'''

Here apt-get install python-xmltodict helped.

Traceback (most recent call last):
  File "/home/mechanik/dev/OpenDroneMap/src/OpenSfM/bin/detect_features", line 11, in <module>
    from opensfm import features
  File "/home/mechanik/dev/OpenDroneMap/src/OpenSfM/opensfm/features.py", line 12, in <module>
    import csfm
ImportError: No module named csfm
b

quitting cause: 
	PYTHONPATH=/home/mechanik/dev/OpenDroneMap/lib/python2.7/dist-packages "/home/mechanik/dev/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.

But here I don't know what to do ;)  libcsfm-dev doesn't help.

dakotabenjamin, 2015-10-28:
I've never tried to install ODM on 15.10, so that may be the issue, but I think it is more likely to be an install issue. csfm.cc should be in src/OpenSfM/opensfm/src/, so check your install logs (logs/opensfm_1_build.log) and rerun install.sh

merkato, 2015-10-28:
Hmm.. csfm.cc is on own place, but I haven't logs/opensfm_1_build.log

I will rerun install.sh

merkato, 2015-10-28:
OpenSfm doesn't built because ExtractUTM not ended correctly. 

make: *** [all] Błąd 2
Makefile:75: polecenia dla obiektu 'all' nie powiodły się
make[1]: *** [CMakeFiles/odm_extract_utm.dir/all] Błąd 2
CMakeFiles/Makefile2:60: polecenia dla obiektu 'CMakeFiles/odm_extract_utm.dir/all' nie powiodły się
make[2]: *** Brak reguł do zrobienia obiektu '/usr/lib/libproj.so', wymaganego przez 'odm_extract_utm'. Stop.
[100%] Building CXX object CMakeFiles/odm_extract_utm.dir/src/UtmExtractor.cpp.o
[ 66%] Building CXX object CMakeFiles/odm_extract_utm.dir/src/Logger.cpp.o
[ 33%] Building CXX object CMakeFiles/odm_extract_utm.dir/src/main.cpp.o
Scanning dependencies of target odm_extract_utm

merkato, 2015-10-28:
$ sudo locate libproj.so
/usr/lib/x86_64-linux-gnu/libproj.so.9
/usr/lib/x86_64-linux-gnu/libproj.so.9.0.0

dakotabenjamin, 2015-10-28:
$ sudo locate libproj.so
/usr/lib/libproj.so
/usr/lib/libproj.so.0
/usr/lib/libproj.so.0.7.0

on my machine

merkato, 2015-10-29:
Finnally...

For Ubuntu 15.10 
'''sudo apt-get install python-xmltodict
sudo ln -s /usr/lib/x86_64-linux-gnu/libproj.so.9 /usr/lib/libproj.so'''
**before** ./install.sh should help.

closed 2015-10-29 by merkato

dakotabenjamin, 2015-10-29:
I just ran into this issue on 14.04....

w0rcy, 2016-03-22:
I am having this same "No module named csfm" issue on 14.04.4. I didn't know it updated past 14.04 until now, but it is a fresh Ubuntu installation. I do get the same as dakotabenjamin for 

$ sudo locate libproj.so
/usr/lib/libproj.so
/usr/lib/libproj.so.0
/usr/lib/libproj.so.0.7.0

I tried:
pip install xmltodict
but got:
Requirement already satisfied (use --upgrade to upgrade): xmltodict in /usr/local/lib/python2.7/dist-packages
Cleaning up...
Suggestions for something to try?

dakotabenjamin, 2016-03-23:
You can try manually installing OpenSfM. I've heard that works.

rajsaurav1234, 2018-04-24:
bash.sh

export PYTHONPATH=$PYTHONPATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages
export PYTHONPATH=$PYTHONPATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/install/lib

error -
Traceback (most recent call last):
  File "run.py", line 11, in <module>
    from scripts.odm_app import ODMApp
  File "/home/ubuntu/IIRS/OpenDroneMap/scripts/odm_app.py", line 5, in <module>
    from opendm import types
  File "/home/ubuntu/IIRS/OpenDroneMap/opendm/types.py", line 5, in <module>
    from opensfm.exif import sensor_string
ImportError: No module named opensfm.exif