OpenDroneMap/ODM #191: How to create a LOD from the large scale point cloud

SimFaris, 2015-10-29:
Now, I have got a large area point cloud from many aerial imagies. ODM can texture the terrain model with the point cloud, but the model is large and the frame is so slow. So My question is how to create the LOD of terrain model from the so large terrain point cloud.Thank you.

smathermather, 2015-11-01:
Ah, good question, @SimFaris. @hobu -- what do you recommend for LOD stuff these days. (BTW, got your e-mail. I'm excited to look at this this weekend.)

mojovski, 2015-11-01:
A system I know about is the po buffer for x3dom:http://x3dom.org/pop/
Loos very promising but 
I didnt try it so far. Would be great to see other opinions here.

smathermather, 2015-11-01:
Nice that it handles textured meshes too.

SimFaris, 2015-11-03:
thank you. I just known that outofcore module of PCL can generate the multi-resolution and out-of-core point cloud based on octree. Maybe I could get the poisson mesh and texture the models on the basis of the every multi-resolution point cloud. But the question will occur because of the skirt. Maybe some can give some more advise.

smathermather, 2015-11-23:
Hi @SimFaris -- any luck with LOD?

SimFaris, 2015-11-27:
I just divided the point cloud with quadtree, and then texture the models with the multi-redolution point cloud. But I did not constrain the points between the neighboring partiotions with the poisson mesh, so the neighboring partiotions are not corresponded.

closed 2016-10-12 by dakotabenjamin