OpenDroneMap/ODM #193: Does OpenDroneMap work on Debian?

cLupus, 2015-10-30:
Hi!
I've been trying for a while to install ODM on a debian machine (Linux 3.2.0-4-amd64 #1 SMP Debian 3.2.65-1 x86_64), but the install script stops installing required packages (updating). The content of the log files can be found at http://server.nistad.me/logs/.

dakotabenjamin, 2015-10-30:
Can you run `sudo apt-get update` from the command line successfully? We've only been testing on Ubuntu so far.

cLupus, 2015-10-30:
Except for a few non-standard sources can’t be found, it runs successfully.

Sincerely yours,
Sindre Nistad

cLupus, 2015-10-30:
Actually, I removed the package that caused the apt-get update command to give a little error. This gets me a little bit further, but now I get a new error:
curl: (77) error setting certificate verify locations:
  CAfile: /etc/pki/tls/certs/ca-bundle.crt
  CApath: none

Sincerely yours,
Sindre Nistad

On October 30, 2015 at 16:06:45, Sindre Nistad (sindre_123@msn.com) wrote:

dakotabenjamin, 2015-10-30:
You may be missing a package:
```
$ sudo apt-get install ca-certificates
```

cLupus, 2015-10-30:
The problem seemed to be that the PointCloudLibrary (https://github.com/PointCloudLibrary/pcl/archive/pcl-1.7.2.tar.gz) will redirect you to a new URL (https://codeload.github.com/PointCloudLibrary/pcl/tar.gz/pcl-1.7.2). curl then only downloads the link to redirection.

Sincerely yours,
Sindre Nistad

smathermather, 2015-11-23:
Hopefully some of this gets easier with the cmake rebuild. Just added enhancement and help wanted tags. How is it going @cLupus?

yjmenezes, 2015-12-09:
@cLupus 

I worked a little bit on install.sh for Debian 7.9 Wheezy 64bits and post into issue #208.
I install into Debian 8.2 x64 too.

take a look, maybe it helps you.

[]s
julio

closed 2016-10-12 by dakotabenjamin