OpenDroneMap/ODM #194: python-port Found 0 usable images

edgarriba, 2015-10-30:
Hi,

trying to tun the pipeline with images without metadata.
Got the above error. Thanks in advance.

---------------
 python ../software/OpenDroneMap-dakotabenjamin/run.py --use-opensfm True

  - configuration:
{'start_with': 'resize', 'end_with': 'odm_orthophoto', 'force_focal': None, 'odm_texturing_textureWithSize': 3600, 'matcher_threshold': 2.0, 'cmvs_maxImages': 500, 'zip_results': False, 'pmvs_threshold': 0.7, 'matcher_preselect': True, 'resize_to': 2400, 'odm_meshing_solverDivide': 9, 'use_opensfm': True, 'pmvs_wsize': 7, 'odm_texturing_textureResolution': 4096, 'matcher_ratio': 0.6, 'odm_meshing_samplesPerNode': 1, 'odm_meshing_maxVertexCount': 100000, 'pmvs_level': 1, 'run_only': None, 'odm_meshing_octreeDepth': 9, 'pmvs_csize': 2, 'matcher_kDistance': 20, 'odm_georeferencing_useGcp': False, 'odm_georeferencing_gcpFile': 'gcp_list.txt', 'matcher_useKnn': True, 'pmvs_minImageNum': 3, 'force_ccd': None}

  - source files - Fri Oct 30 14:50:35  2015

    found 0 usable images

  using max image size of 2400 x 2400

  - preparing images - Fri Oct 30 14:50:35  2015

  - running OpenSfM - Fri Oct 30 14:50:35  2015
running import_bundler
running PYTHONPATH=/home/vagrant/software/OpenDroneMap-dakotabenjamin/lib/python2.7/dist-packages "/home/vagrant/software/OpenDroneMap-dakotabenjamin/src/OpenSfM/bin/import_bundler" opensfm --list list.txt
libdc1394 error: Failed to initialize libdc1394
No bundle.out given.  Importing only the list of images list.txt.
Converting output from Bundler to OpenSfM
b
running PYTHONPATH=/home/vagrant/software/OpenDroneMap-dakotabenjamin/lib/python2.7/dist-packages "/home/vagrant/software/OpenDroneMap-dakotabenjamin/src/OpenSfM/bin/run_all" opensfm
libdc1394 error: Failed to initialize libdc1394
Extracting focal lengths for image 
Traceback (most recent call last):
  File "/home/vagrant/software/OpenDroneMap-dakotabenjamin/src/OpenSfM/bin/focal_from_exif", line 49, in <module>
    d = exif.extract_exif_from_file(data.load_image(image))
  File "/home/vagrant/software/OpenDroneMap-dakotabenjamin/src/OpenSfM/opensfm/dataset.py", line 63, in load_image
    return open(self.__image_file(image))
IOError: [Errno 21] Is a directory: 'opensfm/'
b

quitting cause: 
	PYTHONPATH=/home/vagrant/software/OpenDroneMap-dakotabenjamin/lib/python2.7/dist-packages "/home/vagrant/software/OpenDroneMap-dakotabenjamin/src/OpenSfM/bin/run_all" opensfm
returned with code 256.

dakotabenjamin, 2015-10-30:
When preparing the images for reconstruction, run.py runs the shell command 'jhead' to parse width, height, focal length, and ccd width from the EXIF. These are necessary for the program to function.

edgarriba, 2015-10-30:
right. Is it possible to set this values by hand?

dakotabenjamin, 2015-10-30:
Ah. Currently only support JPG, but looking to improve that in the future.

edgarriba, 2015-10-30:
ok. Thanks!

closed 2015-10-30 by edgarriba