OpenDroneMap/ODM #197: Error code 35072

edgarriba, 2015-11-05:
Hi!

got the following error when running run.pl and run.py on odm_test/benchmark. Any idea?

    quitting cause: 
	"/home/vagrant/odm_test/benchmark/../../software/OpenDroneMap-dakotabenjamin/bin/parallel" --halt-on-error 1 -j+0 < "/home/vagrant/odm_test/benchmark/reconstruction-with-image-size-2400/_vlsift.templist.txt"
    returned with code 35072

dakotabenjamin, 2015-11-05:
try `--use-opensfm True`

edgarriba, 2015-11-05:
libdc1394 error: Failed to initialize libdc1394

edgarriba, 2015-11-05:
ImportError: /home/vagrant/software/OpenDroneMap-dakotabenjamin/src/OpenSfM/opensfm/csfm.so: undefined symbol: _ZTVN2cv12_OutputArrayE
b

dakotabenjamin, 2015-11-05:
That looks like an install error. Check your opensfm install logs.

dakotabenjamin, 2015-11-05:
Or make sure you've pulled the latest version of `python-port` and reinstall.

edgarriba, 2015-11-09:
@dakotabenjamin update and installation succeed correctly on python-port. 
Got the same error in OpenSfM. @paulinus  any idea?

   running OpenSfM - Mon Nov 09 09:47:41  2015
running import_bundler
running PYTHONPATH=/home/vagrant/software/OpenDroneMap/lib/python2.7/dist-packages "/home/vagrant/software/OpenDroneMap/src/OpenSfM/bin/import_bundler" opensfm --list list.txt
libdc1394 error: Failed to initialize libdc1394
No bundle.out given.  Importing only the list of images list.txt.
Converting output from Bundler to OpenSfM
b
running PYTHONPATH=/home/vagrant/software/OpenDroneMap/lib/python2.7/dist-packages "/home/vagrant/software/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
libdc1394 error: Failed to initialize libdc1394
Extracting focal lengths for image DSC05507.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05508.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05509.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05510.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05511.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05512.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05513.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05514.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05515.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05516.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05517.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05518.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05519.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05520.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05521.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05522.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05523.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05524.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05525.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05526.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05527.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05528.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05529.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05530.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
Extracting focal lengths for image DSC05531.JPG
Warning: Missing focal length in EXIF. Using default focal ratio 0.85
libdc1394 error: Failed to initialize libdc1394
Traceback (most recent call last):
  File "/home/vagrant/software/OpenDroneMap/src/OpenSfM/bin/detect_features", line 11, in <module>
    from opensfm import features
  File "/home/vagrant/software/OpenDroneMap/src/OpenSfM/opensfm/features.py", line 12, in <module>
    import csfm
ImportError: /home/vagrant/software/OpenDroneMap/src/OpenSfM/opensfm/csfm.so: undefined symbol: _ZTVN2cv12_OutputArrayE
b

quitting cause: 
	PYTHONPATH=/home/vagrant/software/OpenDroneMap/lib/python2.7/dist-packages "/home/vagrant/software/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.

paulinus, 2015-11-09:
@edgarriba this seems to be a linking problem. It is probably due to the fact that we were installing opencv from source.  Could you try out this branch https://github.com/paulinus/OpenDroneMap/tree/update-opensfm which does use ubuntu's opencv?

edgarriba, 2015-11-09:
@paulinus not working. Same issue that @dakotabenjamin fixed in 3df10ee2f20a8aacf5d0ed5dbbd769471236ba8d

Scanning dependencies of target odm_texturing
[ 50%] [ 75%] [100%] Building CXX object CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o
Building CXX object CMakeFiles/odm_texturing.dir/src/main.cpp.o
Building CXX object CMakeFiles/odm_texturing.dir/src/modifiedPclFunctions.cpp.o
Building CXX object CMakeFiles/odm_texturing.dir/src/Logger.cpp.o
Linking CXX executable odm_texturing
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::loadCameras()':
OdmTexturing.cpp:(.text+0x3812): undefined reference to `cv::imread(cv::String const&, int)'
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::createTextures()':
OdmTexturing.cpp:(.text+0x782e): undefined reference to `cv::imread(cv::String const&, int)'
OdmTexturing.cpp:(.text+0x7dd4): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
OdmTexturing.cpp:(.text+0x805b): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [odm_texturing] Error 1
make[1]: *** [CMakeFiles/odm_texturing.dir/all] Error 2
make: *** [all] Error 2

edgarriba, 2015-11-11:
@dakotabenjamin @paulinus I would say that the linking error comes from the fact that in odm we are forcing to compile with a concrete OpenCV version (2.8) and in OpenSfM there's no restriction regarding OpenCV's version. From my point of view the only solution I see here is re-factoring the installation procedure in order to be aware of this version problems and have a more "standard" process using fully cmake environment to configure the project before compiling.

I opened an issue where we can discuss this question.
https://github.com/OpenDroneMap/OpenDroneMap/issues/201

closed 2016-10-12 by dakotabenjamin