OpenDroneMap/ODM #198: Help about the OpenSFM

xialang2012, 2015-11-05:
Hi, The data I used to obtain the orthodox images can be downloaded from this link. http://xialang2012-img.stor.sinaapp.com/jl3.tar.gz
I run the command "run.py --use-opensfm=true", but I get an low quality image

[image]
but it should like this 
[image]

I tried another test data at http://xialang2012-img.stor.sinaapp.com/7.2.zip with the same command, and it also failed with this message:

>Preemptive matching 0, time: 0.0018470287323s
Discarding based of preemptive matches 0 < 5
Preemptive matching 1, time: 0.00200009346008s
Discarding based of preemptive matches 0 < 5
Discarding based of preemptive matches 1 < 5
Preemptive matching 0, time: 0.00184679031372s
Preemptive matching 0, time: 0.00577712059021s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00188994407654s
Discarding based of preemptive matches 0 < 5
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00200200080872s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00181102752686s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00182700157166s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00182104110718s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00191497802734s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00190401077271s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00179505348206s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00215601921082s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00156283378601s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00160908699036s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00179195404053s
Discarding based of preemptive matches 0 < 5
Preemptive matching 0, time: 0.00192403793335s
Discarding based of preemptive matches 0 < 5
2015-11-04 23:40:12,809 reading features
2015-11-04 23:40:17,166 Merging features onto tracks
2015-11-04 23:40:17,178 Good tracks: 646
images 10
nonfisheye images 10
0 partial reconstructions in total.
Traceback (most recent call last):
  File "/home/xl/open/OpenDroneMap/src/OpenSfM/bin/mesh", line 23, in <module>
    reconstructions = data.load_reconstruction()
  File "/home/xl/open/OpenDroneMap/src/OpenSfM/opensfm/dataset.py", line 292, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: 'opensfm/reconstruction.json'
b

>quitting cause: 
	PYTHONPATH=/home/xl/open/OpenDroneMap/lib/python2.7/dist-packages "/home/xl/open/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.

 I do not know if I need to change the OpenSFM parameters which used in the ODM.
By the way, I can obtained the correct orthodox images by using Pix4d, PhotoScan or ODM with command "run.py --matcher-threshold=0.5 --matcher-ratio=1.0" for test data jl3.tar.gz.

dakotabenjamin, 2015-11-05:
The proper way to pass parameters is as follows: `--param value`. For example, for your data:
```
run.py --use-opensfm True
```
or 
```
run.py --matcher-threshold 0.5 --matcher-ratio 1.0
```

smathermather, 2015-11-23:
Hi @xialang2012 / @dakotabenjamin. Should this issue be closed?

smathermather, 2015-11-23:
I see there's also #204. This should be resolved in https://github.com/OpenDroneMap/OpenDroneMap/pull/203

dakotabenjamin, 2015-11-23:
@edgarriba you got a similar error in your cmake branch, right?

edgarriba, 2015-11-24:
right. The point is that in the original OpenSfM the command "run_all" expects the directory where the images are placed. In here, you pass an argument called "opensfm". Not sure how the branch odm2 works ... because I got the same runtime error. Maybe @paulinus  could help us in that.

paulinus, 2015-11-24:
The problem here seems to be that the reconstruction is failing, probably because of too few matches. I will give a try on those images an let you know how it goes.

This problem is different from https://github.com/edgarriba/OpenDroneMap/issues/5

paulinus, 2015-11-24:
I got this, which looks ok, running opensfm from outside ODM,
[image]
Will try running the complete ODM pipeline latter and report.

paulinus, 2015-11-24:
Here's the orthophoto generated by ODM using the command
```python /vagrant/OpenDroneMap/run.py --use-opensfm --resize-to 1024```
[image]
Not sure what went wrong for you @xialang2012. I'm closing the issue since this dataset seems to work fine. Please re-open it if doesn't work for you.

closed 2015-11-24 by paulinus