OpenDroneMap/ODM #200: odm_orthophoto resolution parameter

lupus78, 2015-11-10:
odm_orthophoto has a parameter "-resolution", which is 20.0pixels/m by default, backed into run.py/pl

This should be a paramter of run.py (eg.: --odm_orthophoto-resolution).
I'm testing with a dataset of traditional aerial images, covering an 8.2x8.2km area, and with 20.0 resolution, the image size would be: 165408x165128
Obviously this causes an out of memory error:
```
  - running orthophoto generation - Tue Nov 10 12:35:36  2015
running "/path/to/code/OpenDroneMap/bin/odm_orthophoto" -inputFile /path/to/data/reconstruction-with-image-size-400-results/odm_texturing/odm_textured_model_geo.obj -logFile /path/to/data/reconstruction-with-image-size-400/odm_orthophoto/odm_orthophoto_log.txt -outputFile /path/to/data/reconstruction-with-image-size-400-results/odm_orthphoto.png -resolution 20.0 -outputCornerFile /path/to/data/reconstruction-with-image-size-400/odm_orthphoto_corners.txt
OpenCV Error: Insufficient memory (Failed to allocate 109253968900 bytes) in OutOfMemoryError, file /path/to/code/OpenDroneMap/src/opencv/modules/core/src/alloc.cpp, line 52
Error in OdmOrthoPhoto:
/path/to/code/OpenDroneMap/src/opencv/modules/core/src/alloc.cpp:52: error: (-4) Failed to allocate 109253968900 bytes in function OutOfMemoryError
```

I had to change run.py to 1.0 resolution to get any result.

Here's the odm_orthophoto_log.txt
```
/path/to/code/OpenDroneMap/bin/odm_orthophoto_log.txt

Arguments given
-inputFile
/path/to/data/reconstruction-with-image-size-400-results/odm_texturing/odm_textured_model_geo.obj
-logFile
/path/to/data/reconstruction-with-image-size-400/odm_orthophoto/odm_orthophoto_log.txt
-outputFile
/path/to/data/reconstruction-with-image-size-400-results/odm_orthphoto.png
-resolution
20.0
-outputCornerFile
/path/to/data/reconstruction-with-image-size-400/odm_orthphoto_corners.txt

Reading textured mesh from: /path/to/data/reconstruction-with-image-size-400-results/odm_texturing/odm_textured_model_geo.obj
Log file path was set to: /path/to/data/reconstruction-with-image-size-400/odm_orthophoto/odm_orthophoto_log.txt
Writing output to: /path/to/data/reconstruction-with-image-size-400-results/odm_orthphoto.png
Resolution count was set to: 20pixels/meter
Writing corners to: /path/to/data/reconstruction-with-image-size-400/odm_orthphoto_corners.txt

Reading mesh file...
.. mesh file read.

Number of faces in the model 113009
Set boundary to contain entire model.
Local boundary points:
Point 1: -5979.7 -5136.95
Point 2: -5979.7 3119.44
Point 3: 2290.69 3119.44
Point 4: 2290.69 -5136.95

Ortho photo bounds x : -5979.7 -> 2290.69
Ortho photo bounds y : -5136.95 -> 3119.44
Ortho photo area : 6.82835e+07m2
Ortho photo resolution, width x height : 165408x165128
Error in OdmOrthoPhoto:
/path/to/code/OpenDroneMap/src/opencv/modules/core/src/alloc.cpp:52: error: (-4) Failed to allocate 109253968900 bytes in function OutOfMemoryError

```

lupus78, 2015-11-10:
Shall I submit a pull request? Are you going to switch to the Python port soon? Is it necessary to update the Perl run.pl too?

dakotabenjamin, 2015-11-10:
We will be merging the python port once we squash the last few bugs, so don't bother with the perl script. This sounds like a great feature to add!

closed 2015-11-12 by dakotabenjamin