OpenDroneMap/ODM #201: CMakeLists.txt for installation

edgarriba, 2015-11-10:
Hi folks,

has anyone considered to port the installation.sh to a CMake? Until now I haven't succeed with installation and run a simple demo from odm_data due to library linking errors. Maybe by doing a standard installation using CMakeLists and submodules could help in addition to easily debug what's going on during the whole process.

smathermather, 2015-11-10:
Definitely interested in make / cmake install for it, especially any work that gets us closer to tying into package management as well.

edgarriba, 2015-11-11:
@smathermather I started a SuperBuild cmake project. 
@dakotabenjamin @paulinus You might be interest in that.
You can check here:
 https://github.com/edgarriba/OpenDroneMap/tree/fd338c40c39c0120e5a4c0d0134fbd339f1ef26e/SuperBuild

paulinus, 2015-11-11:
looking good!

closed 2016-09-12 by dakotabenjamin