OpenDroneMap/ODM #204: T/F parameters always set to True

dakotabenjamin, 2015-11-12:
run the following: 
```
vagrant@vagrant-ubuntu-trusty-64:/vagrant_data/odm_data/seneca$ /odm/run.py --resize-to 1600 -e cmvs --use-opensfm Fals
e --odm_georeferencing-useGcp False

  - configuration:
{'start_with': 'resize', 'end_with': 'cmvs', 'force_focal': None, 'odm_texturing_textureWithSize': 3600, 'matcher_thresh
old': 2.0, 'cmvs_maxImages': 500, 'zip_results': False, 'odm_orthophoto_resolution': 20.0, 'pmvs_threshold': 0.7, 'match
er_preselect': False, 'resize_to': 1600, 'matcher_k': 8, 'odm_meshing_solverDivide': 9, 'min_num_features': 4000, 'use_o
pensfm': True, 'pmvs_wsize': 7, 'odm_texturing_textureResolution': 4096, 'matcher_ratio': 0.6, 'odm_meshing_samplesPerNo
de': 1, 'odm_meshing_maxVertexCount': 100000, 'pmvs_level': 1, 'run_only': None, 'odm_meshing_octreeDepth': 9, 'pmvs_csi
ze': 2, 'matcher_kDistance': 20, 'odm_georeferencing_useGcp': True, 'odm_georeferencing_gcpFile': 'gcp_list.txt', 'match
er_useKnn': True, 'pmvs_minImageNum': 3, 'force_ccd': None}
```

No matter what input, the boolean parameters always set to True when called.

lupus78, 2015-11-12:
Seams like there is already the correct solution in the code:

--zip-results works as if added, means True, otherwise False. I think this should be applied for --use-opensfm and --odm_georeferencing-useGcp
```
parser.add_argument('--zip-results',
                    action='store_true',
                    default=False,
                    help='compress the results using gunzip')

parser.add_argument('--use-opensfm',
                    type=bool,
                    default=False,
                    help='use OpenSfM instead of Bundler to find the camera positions '
                         '(replaces getKeypoints, match and bundler steps)')
```

lupus78, 2015-11-13:
I have just made a pull request with a possible solution.

Looks like argparse don't support the --parameter True/False format. What you can do is make a "--use-xyz" and "--dont-use-xyz" parameters.

I did this for the --use-opensfm and --use-bundler parameters. They are linked, only 1 can specified at a time and opensfm is the default.

With "--odm_georeferencing-useGcp" I did just like the "--zip-results". If you add these it will mean True, otherwise False.

There are 2 more boolean arguments, which I didn't changed because I'm not sure what they do exactly:
--matcher-preselect
--matcher-useKnn

We shall think about what to do with those.

dakotabenjamin, 2015-11-13:
We want to make OpenSfM the default over bundler, but I haven't gotten around to removing bundler completely. We could change the argument to `--use-bundler` with the action `store_false` as a kludge approach for now. 

`--matcher-useKnn` if true, will run a k-nearest neighbor using the exif data to speed up the `matcher` part of the toolchain. 

`--matcher-preselect` is the same thing but for OpenSfM, I believe.

lupus78, 2015-11-14:
My pull request dose this approach with the `--use-bundler` argument. It also has the `--use-opensfm` argument, but it don't need to be specified because this is the default value.

Do you want to completly remove the Bundler solution? I get good results with my none drone images with Bundler... I would keep it.

Tracked down the `--matcher-****` arguments:

`--matcher-preselect` Defaults to False. Use GPS data if available with the Knn or Distance method

`--matcher-useKnn` Defaults to True, used only if `--matcher-preselect` is True. Use the Knn method instead of the distance method.

`--matcher-kDistance`  Distance threshold

`--matcher-k` K value for the k-nearest neighbors method.

in `def match()` only `--matcher-kDistance` is used, and either means K or Distance depending on `--matcher-useKnn`

in `def opensfm()` only the `--matcher-k` is used and always assumed to the Knn method if `--matcher-preselect` is True.

As I see OpenSFM could use the Distance method, or actually it can use both at the same time. https://github.com/mapillary/OpenSfM/blob/master/config.yaml#L46-L47

I think that 4 arguments have to be simplified into 2 args this:

`--matcher-distance <number>` the distance threshold
`--matcher-neighbors <number>' the k nearest neighbors threshold

providing 0 for the arguments would mean don't use the given method. providing value for both would mean to use them together (just like OpenSFM works).

Do you think this could work?

dakotabenjamin, 2015-11-16:
I like how you've written the bundler/OpenSfM arguments, it looks very good. The reason we want to depreciate bundler is twofold: OpenSfM is actively maintained and runs faster than bundler. It is more effective to keep track of only one toolchain, especially when the results from both are nearly the same. 

I think you have surmised that those parameter were developed separately, one by the Spotscale folks (who contributed the python port), and the other by @paulinus. I think if they can be merged, they should be. We already have a lengthy list of params which could be a source of confusion for some. This seems like it would be trivial to do for the bundler side, but I'm not as familiar with OpenSfM as @paulinus.

lupus78, 2015-11-16:
Ok, I'll make these changes and push them.
I agree with your statement on the arguments, they should be cleaned up. I'll review them and adjust the help texts, and display the default values, so `run.py -h` will give a nice output.

closed 2015-12-02 by dakotabenjamin