OpenDroneMap/ODM #208: Compatibility with Ubuntu 12.04

dakotabenjamin, 2015-11-20:
I'm working on the branch called `pp-precise-comp` in order to improve compatibility with Ubuntu 12.04, which is supported until 2017. It would appear that there are a few issues with this:

  * Ceres (and as such, OpenSfM) does not want to compile. This was temporarily going to be solved by disabling ceres and OpenSfM, and forcing the use of bundler instead. 
  * However, bundler apparently ALSO uses ceres (which is good and all, but I can't get it to compile!)

Any help would be appreciated.

dakotabenjamin, 2015-11-24:
Here is my kludge approach to getting it working in 12.04 NOTE OpenSfM does not work; you must specify `--use-bundler` in the run command. 

```

#cmake
curl --progress-bar --location -o cmake-3.4.0.tar.gz https://cmake.org/files/v3.4/cmake-3.4.0.tar.gz
tar xvf cmake-3.4.0.tar.gz
cd cmake-3.4.0
./bootstrap && make && sudo make install
cd 

curl --progress-bar --location -o gflags.tar.gz https://github.com/gflags/gflags/archive/v2.0.tar.gz
tar xvf gflags.tar.gz
cd gflags-2.0
./configure && make && make install
cd

#install google glog:
curl --progress-bar --location -o glog.tar.gz https://github.com/google/glog/archive/v0.3.4.tar.gz
tar xvf glog.tar.gz
cd glog-0.3.4
./configure
make
sudo make install
cd

#compile suitesparse:
curl --progress-bar --location -o suitesparse.tar.gz http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-4.2.1.tar.gz
tar xvf suitesparse.tar.gz
cd SuiteSparse/
make
sudo make install
cd

# Eigen 3:
curl --progress-bar --location -o eigen3.tar.gz https://bitbucket.org/eigen/eigen/get/3.2.0.tar.gz
tar xvf eigen3.tar.gz
cd eigen-eigen-ffa86ffb5570/
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/usr
sudo make install
cd 

# Replace ln 117-130 in install.sh with:
sudo apt-get install --assume-yes --install-recommends \
  build-essential g++ gcc gFortran perl git autoconf \
  curl wget \
  unzip \
  imagemagick jhead proj-bin libproj-dev\
  libjpeg-dev libboost1.46-dev libgsl0-dev libx11-dev libxext-dev liblapack-dev \
  libflann-dev libvtk5-dev libqhull-dev libusb-1.0-0-dev\
  libzip-dev \
  libswitch-perl libjson-perl \
  libcv-dev libcvaux-dev libopencv-dev \
  gdal-bin \
  exiv2 \
  libatlas-base-dev \
  python-dev python-pip \
  python-numpy-dev python-scipy python-yaml \
  python-opencv python-pyexiv2 \
  > "$TOOLS_LOG_PATH/apt-get_install.log" 2>&1
```

dakotabenjamin, 2015-11-30:
@paulinus why do you think OpenSfM doesn't work under 12.04 and what steps do you think I should take to get it functional? It compiles alright but running it runs into this error:
```
Traceback (most recent call last):
  File "/home/vagrant/OpenDroneMap/src/OpenSfM/bin/detect_features", line 11, in <module>
    from opensfm import features
  File "/home/vagrant/OpenDroneMap/src/OpenSfM/opensfm/features.py", line 12, in <module>
    import csfm
ImportError: /home/vagrant/OpenDroneMap/src/OpenSfM/opensfm/csfm.so: undefined symbol: cs_di_chol
b

quitting cause:
        PYTHONPATH=/home/vagrant/OpenDroneMap/lib/python2.7/dist-packages "/home/vagrant/OpenDroneMap/src/OpenSfM/bin/ru
n_all" opensfm
returned with code 256.
```
The only issue I found in OpenSfM was [#41](https://github.com/mapillary/OpenSfM/issues/41). Was there any resolution? Do I need to fix to libsuitesparse build?

paulinus, 2015-12-01:
This seems a problem with ceres and its dependencies. I'm trying to reproduce it. Will let you know when i get something.

dakotabenjamin, 2015-12-02:
Updated: 
```

#cmake
curl --progress-bar --location -o cmake-3.4.0.tar.gz https://cmake.org/files/v3.4/cmake-3.4.0.tar.gz
tar xvf cmake-3.4.0.tar.gz
cd cmake-3.4.0
./bootstrap && make && sudo make install
cd 

curl --progress-bar --location -o gflags.tar.gz https://github.com/gflags/gflags/archive/v2.0.tar.gz
tar xvf gflags.tar.gz
cd gflags-2.0
./configure && make && make install
cd

#install google glog:
curl --progress-bar --location -o glog.tar.gz https://github.com/google/glog/archive/v0.3.4.tar.gz
tar xvf glog.tar.gz
cd glog-0.3.4
./configure
make
sudo make install
cd

#compile suitesparse:
curl --progress-bar --location -o suitesparse.tar.gz http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-4.2.1.tar.gz
tar xvf suitesparse.tar.gz
cd SuiteSparse/
make
sudo make install
cd

# Eigen 3:
curl --progress-bar --location -o eigen3.tar.gz https://bitbucket.org/eigen/eigen/get/3.2.0.tar.gz
tar xvf eigen3.tar.gz
cd eigen-eigen-ffa86ffb5570/
mkdir build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/usr
sudo make install
cd 

# Opencv
curl --progress-bar --location -o opencv.zip https://github.com/Itseez/opencv/archive/2.4.11.zip
unzip opencv.zip
cd opencv-2.4.11/
mkdir build && cd build
cmake ..
sudo make install

# Replace ln 117-130 in install.sh with:
sudo apt-get install --assume-yes --install-recommends \
  build-essential g++ gcc gFortran perl git autoconf \
  curl wget \
  unzip \
  imagemagick jhead proj-bin libproj-dev\
  libjpeg-dev libboost1.46-dev libgsl0-dev libx11-dev libxext-dev liblapack-dev \
  libflann-dev libvtk5-dev libqhull-dev libusb-1.0-0-dev\
  libzip-dev \
  libswitch-perl libjson-perl \
  libcv-dev libcvaux-dev \
  gdal-bin \
  exiv2 \
  libatlas-base-dev \
  python-dev python-pip \
  python-numpy-dev python-scipy python-yaml \
  python-opencv python-pyexiv2 \
  > "$TOOLS_LOG_PATH/apt-get_install.log" 2>&1
```

yjmenezes, 2015-12-03:
i failed to run install.sh under Debian 7.9, it seems the problem is automake version.
Debian 7.9 automake is 1.11.6 and google glog require 1.14 

I have success running install.sh under Debian 8.2, automake version 1.14.1 

i hope it helps to fix Ubuntu 12.04.

julio menezes

yjmenezes, 2015-12-04:
I got  Script finished under Ubuntu 12.04.5, i hope this build is ok !

Some minor changes:
a) fix missing "sudo" make install ( gflags )
b) install automake 1.15 for glog and ceres
c) replace libboost 1.46 by 1.48

julio menezes
PS: fix1204.sh bellow

#!/bin/bash
# issue 208 fix Ubuntu 12.04.5  ver 1.0 2015-12-04
# git clone https://github.com/OpenDroneMap/OpenDroneMap.git
# https://github.com/OpenDroneMap/OpenDroneMap/issues/208  Add custom glog #2
# TODO pcl require replace libboost 1.46.1 by 1.47 or newer
echo "step0 | automake | boost | cmake | gflags | glog | suitesparces | eigen3 | opencv | help" 1>&2
case $1 in
"step0")
sudo apt-get update

sudo apt-get install --assume-yes --install-recommends build-essential g++ gcc gFortran \
 perl git autoconf  curl wget  unzip  imagemagick jhead proj-bin libproj-dev  \
 libjpeg-dev  libgsl0-dev libx11-dev libxext-dev liblapack-dev \
 libflann-dev libvtk5-dev libqhull-dev libusb-1.0-0-dev libzip-dev  \
 libswitch-perl libjson-perl  \
 gdal-bin  exiv2  libatlas-base-dev \
 python-dev python-pip  python-numpy-dev python-scipy python-yaml  python-pyexiv2 
;;
#                                        
# automake for glog and ceres
"automake")
curl --progress-bar --location -o automake.tar.gz http://ftp.gnu.org/gnu/automake/automake-1.15.tar.gz
tar xzvf automake.tar.gz
cd automake-1.15
./configure
make
sudo make install
automake --version 1>&2
echo "done automake" $(date) 1>&2
cd
;;

"boost")
# ubuntu synaptic :  libboost-all-dev 1.48.0.2
#curl --progress-bar --location -o boost.tar.bz2 http://sourceforge.net/projects/boost/files/boost/1.59.0/boost_1_59_0.tar.bz2
dpkg -S /usr/include/boost/version.hpp
sudo apt-get autoremove libboost1.46-dev
# newer for PCL
sudo apt-get install  libboost1.48-all-dev
dpkg -S /usr/include/boost/version.hpp
#
cd
;;
#cmake
"cmake")
curl --progress-bar --location -o cmake-3.4.0.tar.gz https://cmake.org/files/v3.4/cmake-3.4.0.tar.gz
tar xvf cmake-3.4.0.tar.gz
cd cmake-3.4.0
./bootstrap && make && sudo make install
cd 
;;
"gflags")
curl --progress-bar --location -o gflags.tar.gz https://github.com/gflags/gflags/archive/v2.0.tar.gz
tar xvf gflags.tar.gz
cd gflags-2.0
./configure && make && sudo make install
cd
;;
"glog")
[ -d glog-0.3.4 ] && sudo rm -r glog-0.3.4
#install google glog:
curl --progress-bar --location -o glog.tar.gz https://github.com/google/glog/archive/v0.3.4.tar.gz
tar xvf glog.tar.gz
cd glog-0.3.4
./configure
make
sudo make install
cd
;;
"suitesparses")
#compile suitesparse:
curl --progress-bar --location -o suitesparse.tar.gz http://faculty.cse.tamu.edu/davis/SuiteSparse/SuiteSparse-4.2.1.tar.gz
tar xvf suitesparse.tar.gz
cd SuiteSparse/
make
sudo make install
cd
;;
"eigen3")
# Eigen 3:
curl --progress-bar --location -o eigen3.tar.gz https://bitbucket.org/eigen/eigen/get/3.2.0.tar.gz
tar xvf eigen3.tar.gz
cd eigen-eigen-ffa86ffb5570/
mkdir -p build && cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/usr
sudo make install
cd 
;;
"opencv")
# Opencv
curl --progress-bar --location -o opencv.zip https://github.com/Itseez/opencv/archive/2.4.11.zip
unzip opencv.zip
cd opencv-2.4.11/
mkdir -p build && cd build
cmake ..
sudo make install
cd
;;
"help")
echo "please, select a task" 1>&2
;;
*)
esac
exit 0

# Replace ln 117-130 in install.sh with:
# Please, remove  libboost1.46-dev, replace by newer for PCL

yjmenezes, 2015-12-04:
I apologise, i did not type  bold or upercase.
regards,
julio menezes

yjmenezes, 2015-12-08:
Fixed for Debian 7.9 Wheezy, need tests. 

[]s
julio menezes

[fixdeb7_sh.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/55536/fixdeb7_sh.txt)

smathermather, 2016-12-20:
Are we giving up on 12.04? If so, we should close this issue.

dakotabenjamin, 2016-12-21:
16.04 is the new LTS so yeah

closed 2016-12-21 by dakotabenjamin