OpenDroneMap/ODM #211: odm_orthophoto issues continue

rionlerm, 2015-11-25:
Hi @dakotabenjamin and @smathermather .

So I ./intall.sh to recent .py/.pl merge.

1) When running on large dataset, still memory issues with orthophoto creation as with previous "version" that you are aware of.

2) However, small dataset results in stretched orthophoto. I added no additional runtime parameters and the previous "version" of ODM created a good ortho. See stretched image (most recent output) attached. What I did note is that the good output with previous "version" was of file size ~26MB but, this stretched image is ~7MB in size (PNG files)? The textured model though produced correctly.
[image]

Just as a side-note: this attached ortho's result through mapsmadeeasy was of better quality than previous ODM output.
Specifically, where the photos were merged, colour differences were noted with ODM result (see attached Meshlab 3D snapshot).
[image]

lupus78, 2015-11-26:
Isn't the 1st issue solved with the `--odm_orthophoto-resolution` parameter? It was recently added to the Python port.
https://github.com/OpenDroneMap/OpenDroneMap/issues/200
By default it is 20, which produced a very large image (20 pixels/meter), so if you lower that value you will get the output.

dakotabenjamin, 2015-12-09:
Did @lupus78 solution solve your problem? The weird warping may be caused by the subset of photos you are using. 

Your textured mode looks really great. Note that computer vision in general has trouble with reflective surfaces like water- if it is not very clear water on a sunny or partially cloudy day you may get mixed results.

rionlerm, 2015-12-10:
Hi @dakotabenjamin 

I am actually running the script now with @lupus78 suggestions on LARGE dataset with lower resolution ortho.

Have not changed the subset of photos on SMALL dataset. Will attempt after large and let you know.

rionlerm, 2015-12-10:
LARGE dataset output looks good but, after I forced 

    --zip-results

the ortho TIFF is massive (1.6GB). Mapsmadeeasy one is about 250MB?

rionlerm, 2015-12-10:
OK, thanks to all the geniuses. @dakotabenjamin @lupus78 @smathermather 

Found zipped results way below photos at end of folder contents. SMALL dataset results looks good as well.

rionlerm, 2015-12-10:
Please see comparison between ODM and MapsMadeEasy products.
*Note that both top and bottom maps were aligned to the same canvas extent in GIS so if parts of the river is on or off the map then it was produced by the two producers (ODM and MapsMadeEasy) not GIS-related.*

Latitudinally and Longitudinally ODM comes short with regards to alignment of orthophotos to Earth's surface when overlaid onto Google satellite maps (Google not shown here). Luckily for our work thus far, it is not critical that alignment is accurate. Only product pixel quality and proportions that also seem to differ between the two products, are important for my work.

[image]

dakotabenjamin, 2015-12-10:
Thanks for the comparison! You bring up a good point about alignment- one that I've been struggling with too. Your results are only as good as the GPS used to geolocate the photos. Survey-grade ground control at this time is the best way to go. Fortunately, you can correct the alignment in post-processing. 

In order to improve pixel quality of the orthophoto, you should play around with `--odm_texturing-textureResolution`, `--odm_texturing-textureWithSize`, and `--odm_orthophoto-resolution`.

This will affect the TIFF file size, for better or worse. I would guess that mapsmadeeasy down-samples the images to file transfer.

rionlerm, 2015-12-11:
Thanks @dakotabenjamin , the pointers are much appreciated. I will play with those runtime parameters and start working on GCPs. We have a Leica DGPS I just have to figure out.

closed 2016-10-12 by dakotabenjamin