OpenDroneMap/ODM #213: odm_georef wrong GCP file parsing

lupus78, 2015-11-27:
The gcp_list.txt according to the documentation should be in this format
```
coordinate system description
x1 y1 z1 pixelx1 pixely1 imagename1
```

However if I make my gcp_list.txt like that the georeferencing fails (big mean error, and the orthophoto is rotated ~90deg)

The issue comes from here:
https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/odm_georef/src/Georef.cpp#L31
```
void GeorefGCP::extractGCP(std::istringstream &gcpStream)
{
    gcpStream >> x_ >> y_ >> z_ >> pixelY_ >> pixelX_ >> image_;
}
```
odm_georef reads up the pixel coordinates in the other order (Y, X)

It is not intuitive and should be changed. I would make it, and the change seams easy, however I never coded C... Anyone can fix this?

Fi156, 2015-11-27:
Can confirm this issue.

smathermather, 2015-11-28:
Yup, this looks like a bug. If I understand correctly, it looks like it should be:
```
void GeorefGCP::extractGCP(std::istringstream &gcpStream)
{
    gcpStream >> x_ >> y_ >> z_ >> pixelX_ >> pixelY_ >> image_;
}
```

Go ahead and do a pull request, and we'll test.

smathermather, 2015-11-30:
Ok, fixed in a branch. @dakotabenjamin, do we have a dataset that would work for testing?

https://github.com/OpenDroneMap/OpenDroneMap/pull/215

Fi156, 2015-11-30:
I tested the new branch with my own dataset and compared the resulting orthophoto to an older version.
Without changing the gcp file: The result gets rotated by ~90deg.
With inverted coordinates (x before y) the result looks like the older version.

 This means that the fix worked like expected, at least in my case.

I think that the gcp_list.txt file from the COPR dataset uses the reversed version (y before x).

dakotabenjamin, 2015-11-30:
[Boruszyn_kap](https://github.com/merkato/odm_boruszyn_kap)

merkato, 2015-11-30:
I've added some more ground control points to this dataset. But, someone pointed out that this is not UTM cordinate system (very simillar in concept) but EPSG: 2180 (PL CS1992) National Geodetic Cordinate System, and this will not work.

lupus78, 2015-11-30:
https://github.com/OpenDroneMap/OpenDroneMap/issues/214
This will solve your Coordinate system issue. I'll do a pull request soon.

dakotabenjamin, 2015-11-30:
Sweet. @merkato I submitted a PR for a quick fix of the gcp_list so it is in WGS84. It worked for georeferencing but not for orthophoto generation.

dakotabenjamin, 2015-12-09:
PR was merged.

closed 2015-12-09 by dakotabenjamin