OpenDroneMap/ODM #214: Coordinate system handling

lupus78, 2015-11-27:
Coordinate system handling is very limited in ODM. 

If you are using GCPs, the 1st line of the list file is the Coordinate system, which the wiki say "nothing formal". Actually If you are using the "WGS84 UTM 17N" format it will work, otherwise not. (odm_orthophoto will halt at the geotiff creation)

If no GCP file is used, odm_extract_utm will split out a coordinate file with the same format (WGS84 UTM 17N)

In run.py the above coordinate system string is parsed, and transformed into two formats: PROJ4 and and EPSG number.

The info is used by these commands:
`cs2cs` utility (part of proj4) is used to write the gps exif back to the resized images
`gdal_translate` to make the GeoTIFF
`txt2las` (LASTools) to make the LIDAR files

So what I see is a mess. This dosen't allow any other reference system than WGS UTM xx.

I would stick with GDAL library to make transformations. `cs2cs` could be replaced with `gdaltransform`.
GDAL can understand proj4, EPSG, WKT coordinate system definitions.
These definitions could go into the first line of gcp_list.txt.

Only limitation I found is that `txt2las` only knows a limited set of coordinate systems, so we may need to skip las creation for those unknow systems.

smathermather, 2015-11-27:
Thanks for diving in. Re: txt2las, I'd like to replace that with something from http://www.pdal.io anyway, so that dependency may not rigidly limit our outputs if it's removed.

dakotabenjamin, 2015-12-04:
I created a `georef` feature branch for working on numerous GCP/georeferencing improvements.

closed 2016-10-12 by dakotabenjamin