OpenDroneMap/ODM #217: Error code 256 no module named csfm

DapperFactory, 2015-12-01:
Hello everyone,
Iv been working on getting ODM to run using the python run.py    command within the directory of images. Initially I had some trouble with a few other dependency that went away after downloading and install opensfm. 

Now I get a new error, when trying to import csfm. Is there another way I can install this??

Here is the error:

2015-12-01 12:37:31,469 Extracting focal lengths for image DSC_0817.JPG
Traceback (most recent call last):
  File "/home/wes/Repos/OpenDroneMap/src/OpenSfM/bin/detect_features", line 11, in <module>
    from opensfm import features
  File "/home/wes/Repos/OpenDroneMap/src/OpenSfM/opensfm/features.py", line 12, in <module>
    import csfm
ImportError: No module named csfm
b

quitting cause:
    PYTHONPATH=/home/wes/Repos/OpenDroneMap/lib/python2.7/dist-packages "/home/wes/Repos/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.

Much thanks!!

paulinus, 2015-12-01:
It looks like opensfm has not been build. That is why csfm is missing.

Can you try doing
```
cd /home/wes/Repos/OpenDroneMap/src/OpenSfM
python setup.py build
```

DapperFactory, 2015-12-01:
Thanks for the reply!

So I have already tried that... :(
To clarify, do I need to manually install OpenSfm via the DockerFile. If so what is the proper way to go about this. I ran 
$docker build . 
inside  /home/wes/Repos/OpenDroneMap/src/OpenSfM

which completed without errors. 
And then I ran the
$python setup.py  build

paulinus, 2015-12-01:
You don't need to run docker at all for OpenDroneMap. Just `python setup.py build`. If that completes without errors it will generate the file `src/OpenSfM/opensfm/csfm`, which is the one missing.

Did you run `install.sh` to install OpenDroneMap? It should have done the step above by itself.

DapperFactory, 2015-12-01:
Yes the first thing I ran  was the install.sh script. 

However, when I run it, it does not throw any errors but it does not print "script finished" at the end. 
See below:
Thanks for your help :) !!
sudo ./install.sh 

     created by Daniel Schwarz/daniel.schwarz@topoi.org
     released under Creative Commons/CC-BY
     Attribution

     if the script doesn't finish properly
     (i.e. it doesn't print "script finished" at the end)
     please email me the content of the logs folder

  - script started - Tue Dec  1 12:39:38 EST 2015

  > installing required packages
    - updating
    - installing
The directory '/home/wes/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/wes/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied (use --upgrade to upgrade): networkx in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): exifread in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): xmltodict in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied (use --upgrade to upgrade): decorator>=3.4.0 in /usr/lib/python2.7/dist-packages (from networkx)
    - installing git submodules
Submodule path 'src/bundler': checked out 'b34cc5dcd54345272028b68f82fa4d0ae8a103c9'
  < done - Tue Dec  1 12:39:54 EST 2015

  > getting the sources
    - already downloaded http://ftp.gnu.org/gnu/parallel/parallel-20141022.tar.bz2
    - already downloaded http://www.netlib.org/clapack/clapack-3.2.1-CMAKE.tgz
    - already downloaded http://www.cs.jhu.edu/~misha/Code/PoissonRecon/Version2/PoissonRecon.zip
    - already downloaded http://www.vlfeat.org/download/vlfeat-0.9.13-bin.tar.gz
    - already downloaded http://www.di.ens.fr/cmvs/cmvs-fix2.tar.gz
    - already downloaded http://smathermather.github.io/BundlerTools/patched_files/src/graclus/graclus1.2.tar.gz
    - already downloaded https://github.com/PointCloudLibrary/pcl/archive/pcl-1.7.2.tar.gz
    - already downloaded http://ceres-solver.org/ceres-solver-1.10.0.tar.gz
    - already downloaded http://lastools.org/download/LAStools.zip
Cloning into '/home/wes/Repos/OpenDroneMap/src/opengv'...
remote: Counting objects: 6491, done.
remote: Total 6491 (delta 0), reused 0 (delta 0), pack-reused 6491
Receiving objects: 100% (6491/6491), 6.56 MiB | 6.36 MiB/s, done.
Resolving deltas: 100% (3422/3422), done.
Checking connectivity... done.
Cloning into '/home/wes/Repos/OpenDroneMap/src/OpenSfM'...
remote: Counting objects: 5000, done.
remote: Compressing objects: 100% (11/11), done.
remote: Total 5000 (delta 3), reused 0 (delta 0), pack-reused 4989
Receiving objects: 100% (5000/5000), 18.42 MiB | 6.67 MiB/s, done.
Resolving deltas: 100% (3263/3263), done.
Checking connectivity... done.
  < done - Tue Dec  1 12:39:59 EST 2015

  - unzipping sources
  < done - Tue Dec  1 12:40:03 EST 2015

  - copying patches

  < done - Tue Dec  1 12:40:03 EST 2015

  - building

  > graclus
    - cleaning graclus
    - building graclus
  < done - Tue Dec  1 12:40:11 EST 2015

  > poisson surface reconstruction 
    - building poisson surface reconstruction
  < done - Tue Dec  1 12:40:14 EST 2015

  > parallel
    - configuring parallel
    - building paralel
  < done - Tue Dec  1 12:40:16 EST 2015

  > clapack
    - building clapack
    - installing clapack
  < done - Tue Dec  1 12:40:21 EST 2015

  > vlfeat
    - installing vlfeat
  < done - Tue Dec  1 12:40:21 EST 2015

  > LAStools
    - installing LAStools
  < done - Tue Dec  1 12:40:42 EST 2015

  > cmvs/pmvs
    - cleaning cmvs
    - building cmvs
    - make depend cmvs
  < done - Tue Dec  1 12:40:51 EST 2015

  > ceres
    - configuring ceres
    - building ceres
  < done - Tue Dec  1 12:41:45 EST 2015

  > bundler
    - cleaning bundler
    - building bundler
  < done - Tue Dec  1 12:42:00 EST 2015

  > pcl 
    - configuring pcl
    - building and installing pcl
  < done - Tue Dec  1 12:48:41 EST 2015

  > meshing 
    - configuring odm_meshing
    - building odm_meshing
  < done - Tue Dec  1 12:48:50 EST 2015

  > texturing 
    - configuring odm_texturing
    - building odm_texturing
wes@wes-devbox:~/Repos/OpenDroneMap$ ls

DapperFactory, 2015-12-01:
Looking at the output a second time: 
...
The directory '/home/wes/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/home/wes/.cache/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
...
Could this be what the problem is?

paulinus, 2015-12-01:
From the log above, it seems that texturing build failed. Can you paste the contents of `logs/odm_texturing_2_build.log` here?

DapperFactory, 2015-12-01:
Scanning dependencies of target odm_texturing
[ 25%] [ 50%] [ 75%] [100%] Building CXX object CMakeFiles/odm_texturing.dir/src/main.cpp.o
Building CXX object CMakeFiles/odm_texturing.dir/src/Logger.cpp.o
Building CXX object CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o
Building CXX object CMakeFiles/odm_texturing.dir/src/modifiedPclFunctions.cpp.o
Linking CXX executable odm_texturing
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::loadCameras()':
OdmTexturing.cpp:(.text+0x3812): undefined reference to `cv::imread(cv::String const&, int)'
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In function `OdmTexturing::createTextures()':
OdmTexturing.cpp:(.text+0x782e): undefined reference to `cv::imread(cv::String const&, int)'
OdmTexturing.cpp:(.text+0x7dd4): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
OdmTexturing.cpp:(.text+0x805b): undefined reference to `cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [odm_texturing] Error 1
make[1]: *** [CMakeFiles/odm_texturing.dir/all] Error 2
make: *** [all] Error 2

paulinus, 2015-12-01:
It seems the linker is not finding opencv. It is estrange since install.sh should install it. Are you using ubuntu 14.04?

DapperFactory, 2015-12-01:
Yes I am using ubuntu 14.04 and can successfully import cv2 into python projects. I also had opencv3 installed and running correctly.

Is opencv supposed to reside somewhere specific.

paulinus, 2015-12-01:
install.sh will install ubuntu's opencv package. That puts opencv libs in `/usr/lib/x86_64-linux-gnu`.

It might be happening that the build is mixing opencv2 and opencv3 if you have both installed.

You can tell cmake to look for opencv in a particular folder by modifying the line https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/install.sh#L468
For example,
```
cmake . -DPCL_DIR="$TOOLS_LIB_PATH/pcl" -DOPENCV_DIR="/usr/lib/x86_64-linux-gnu" > "$TOOLS_LOG_PATH/odm_texturing_1_build.log" 2>&1```

DapperFactory, 2015-12-01:
So I should point it to where that is located inside opencv2, correct?

cmake looks like this is all relative on where its looking for dependencies. 

ODM CMAKE:
 ## dest base path
TOOLS_PATH="$PWD"
 ## paths for the tools
      TOOLS_BIN_PATH="$TOOLS_PATH/bin"
      TOOLS_INC_PATH="$TOOLS_PATH/include"
      TOOLS_LIB_PATH="$TOOLS_PATH/lib"
       TOOLS_SRC_PATH="$TOOLS_PATH/src"
      TOOLS_LOG_PATH="$TOOLS_PATH/logs"
      TOOLS_PATCHED_PATH="$TOOLS_PATH/patched_files"

            ## loacal dest paths
		        LIB_PATH="/usr/local/lib"    <-------make sure this points to opencv2?
		        INC_PATH="/usr/local/include"  <-------make sure this points to opencv2?
		
 ODM_TEXTURING_PATH="$TOOLS_SRC_PATH/odm_texturing" <-- this is relative inside ODM

DapperFactory, 2015-12-02:
update:
I added the 
``` -DOPENCV_DIR="/usr/lib/x86_64-linux-gnu" > "$TOOLS_LOG_PATH/odm_texturing_1_build.log" 2>&1 ```

but still have the same error inside the log file. 

Inside the root directory of OpenDroneMap the version of opencv that is called when importing it with python is 2.4.8.

running a grep commands inside /usr/lib/x86_64-linux-gnu/:
Does not seem to return anything related to opencv, however Im not sure if that is an accurate way of determining if the install.sh is in fact install opencv correctly.

DapperFactory, 2015-12-02:
Problem solved & and the morral of the story is do not use OpenCV3

the resulting error:
```undefined reference to cv::imread(cv::String const&, int)'```
seemed to be from "imread was moved to opencv_imgcodecs. Try to add -lopencv_imgcodecs to linker flags."

In any case I tried over and over again to compiled the odm_texture by its self and noticed that even with the correct flag DOPENCV_DIR the error still occurred. 

Solution:
```$cd /ROOT_OPENCV3/build/
$make uninstall ```
Download & install OpenCV-2.4.11

closed 2015-12-02 by DapperFactory

paulinus, 2015-12-02:
different opencv versions are a bit of a pain. We will support opencv3 at some point though.
happy to hear that it works for you!

dakotabenjamin, 2015-12-02:
Referring to #210, I'm going to push towards having more control over the opencv version in ODM. I can work on getting that in this week.

w0rcy, 2016-03-24:
I am having this same "No module named csfm" issue on 14.04.4. This is a fresh Ubuntu installation with a fresh installation of ODM. This must all work if you have all of the history installed as time went by, but I cannot get this to work as it currently exists. Perhaps if I could understand exactly where and what this means to actually do

-Solution:
-$cd /ROOT_OPENCV3/build/ $make uninstall
-Download & install OpenCV-2.4.11

then I could hack a solution in, but this package does not work the way it is. 

Could someone explain this fix a little more and maybe see if the stuff in the package as it exists is actually is what is thought to be there?

dakotabenjamin, 2016-03-24:
If you don't have OpenCV installed, then this issue does not apply to you. I think this issue conflated two separate issues. Would you mind opening a new issue and posting your install logs for opensfm?

profprofgit, 2016-04-10:
u just need to update the CMakeList.txt file as follows:

We need to add absolute path of installed opencv libraries.

target_link_libraries(odm_orthophoto ${PCL_COMMON_LIBRARIES} ${PCL_IO_LIBRARIES} ${PCL_SURFACE_LIBRARIES} ${OpenCV_LIBS} /usr/local/lib/libopencv_core.so.3.1.0 /usr/local/lib/libopencv_core.so.3.1 /usr/local/lib/libopencv_flann.so.3.1.0 /usr/local/lib/libopencv_flann.so.3.1 /usr/local/lib/libopencv_imgproc.so.3.1.0 /usr/local/lib/libopencv_imgproc.so.3.1 /usr/local/lib/libopencv_ml.so.3.1.0 /usr/local/lib/libopencv_ml.so.3.1 /usr/local/lib/libopencv_photo.so.3.1.0 /usr/local/lib/libopencv_photo.so.3.1 /usr/local/lib/libopencv_video.so.3.1.0 /usr/local/lib/libopencv_video.so.3.1 /usr/local/lib/libopencv_imgcodecs.so.3.1.0 /usr/local/lib/libopencv_imgcodecs.so.3.1 /usr/local/lib/libopencv_shape.so.3.1.0 /usr/local/lib/libopencv_shape.so.3.1 /usr/local/lib/libopencv_videoio.so.3.1.0 /usr/local/lib/libopencv_videoio.so.3.1 /usr/local/lib/libopencv_highgui.so.3.1.0 /usr/local/lib/libopencv_highgui.so.3.1 /usr/local/lib/libopencv_objdetect.so.3.1.0 /usr/local/lib/libopencv_objdetect.so.3.1 /usr/local/lib/libopencv_superres.so.3.1.0 /usr/local/lib/libopencv_superres.so.3.1 /usr/local/lib/libopencv_features2d.so.3.1.0 /usr/local/lib/libopencv_features2d.so.3.1 /usr/local/lib/libopencv_calib3d.so.3.1.0 /usr/local/lib/libopencv_calib3d.so.3.1  /usr/local/lib/libopencv_stitching.so.3.1.0 /usr/local/lib/libopencv_stitching.so.3.1 /usr/local/lib/libopencv_videostab.so.3.1.0 /usr/local/lib/libopencv_videostab.so.3.1) 

This will insert all the mentioned depenedant libraries in link.txt file.

Make sure you do this when all tar/zip file gets unzipped in the install.sh file.