OpenDroneMap/ODM #218: bad texturing model

dakotabenjamin, 2015-12-04:
First seen by @edgarriba over [here](https://github.com/edgarriba/OpenDroneMap/issues/8), I've now come across two instances of the bad texturing model. I pulled the `fix1gcpOrder` branch and ran odm_data bellus with a GCP file:
```
WGS84 UTM 17N
441024.15704911412 4564001.8747091573 345.1689312 2648 2147 IMG_1356_RGB.jpg 
441004.67037577706 4564119.5874408539 351.3310728 1678 1442 IMG_1346_RGB.jpg
440934.5930160007 4564245.8292447301 350.638872 1681 1484 IMG_1382_RGB.jpg
440920.08994018572 4564153.5641937079 345.6962352 2213 1267 IMG_1338_RGB.jpg
```
Run command: `run.py --odm_georeferencing-useGcp`
The output:
[image]

One note: ran into an error during texturing and georeferencing which did not kill the process:
`libdc1394 error: Failed to initialize libdc1394`
I tried to remove via apt-get, then reinstall ODM, but it did not go away.

edgarriba, 2015-12-09:
if you remove libdc1394 via apt-get  you need to recompile again opencv

dakotabenjamin, 2015-12-09:
#219 fixes this problem.

dakotabenjamin, 2015-12-09:
As mentioned before, the libdc1394 is a sort of irrelevant issue that can be ignored. It affects the run script in no way.

closed 2016-10-12 by dakotabenjamin