OpenDroneMap/ODM #221: ODM stopped odm-meshing, -input file has a bad value

asierrl, 2015-12-12:
HI, I am new to OpenDroneMap and it really looks great. It is a great jo!! Thanks a lot for sharing it.
I was playing around with some pictures I have. I run it with GCPs from my image folder but it did not finish processing, stopping at the meshing step. The issue is that all the ply files created so far cannot be opened with CloudCompare or Meshlab, which I don't know if it indicates they are no valid. Meshlab reports "Unexpected eof". 
I have also realized that the folders Visualize and txt, within PMVS folder only contained files for 35 out of 120 images.
This is the outuput error in the terminal.

running cp -Rf "/media/DATOS/Activos/Seguimento Illas/Resultados/SeguimentoCabalos/Ortofotos13112015/VueloAldea/Aldea-ODM/reconstruction-with-image-size-2400/pmvs/models" "/media/DATOS/Activos/Seguimento Illas/Resultados/SeguimentoCabalos/Ortofotos13112015/VueloAldea/Aldea-ODM/reconstruction-with-image-size-2400-results"
b

  - running meshing - Sat Dec 12 04:10:07  2015
running "/home/asier/OpenDroneMap/bin/odm_meshing" -inputFile /media/DATOS/Activos/Seguimento Illas/Resultados/SeguimentoCabalos/Ortofotos13112015/VueloAldea/Aldea-ODM/reconstruction-with-image-size-2400-results/option-0000.ply -outputFile /media/DATOS/Activos/Seguimento Illas/Resultados/SeguimentoCabalos/Ortofotos13112015/VueloAldea/Aldea-ODM/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -logFile /media/DATOS/Activos/Seguimento Illas/Resultados/SeguimentoCabalos/Ortofotos13112015/VueloAldea/Aldea-ODM/reconstruction-with-image-size-2400/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9
Error in OdmMeshing:
Argument '-inputFile' has a bad value. (file not accessible)
For more detailed information, see log file.
b

quitting cause: 
	"/home/asier/OpenDroneMap/bin/odm_meshing" -inputFile /media/DATOS/Activos/Seguimento Illas/Resultados/SeguimentoCabalos/Ortofotos13112015/VueloAldea/Aldea-ODM/reconstruction-with-image-size-2400-results/option-0000.ply -outputFile /media/DATOS/Activos/Seguimento Illas/Resultados/SeguimentoCabalos/Ortofotos13112015/VueloAldea/Aldea-ODM/reconstruction-with-image-size-2400-results/odm_mesh-0000.ply -logFile /media/DATOS/Activos/Seguimento Illas/Resultados/SeguimentoCabalos/Ortofotos13112015/VueloAldea/Aldea-ODM/reconstruction-with-image-size-2400/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9
returned with code 256.

Can someone give me a hint on what to try now to fix the problem?
Best regards, 
            Asier

lupus78, 2015-12-12:
I think this may be related with this issue: #64 

You shall not have spaces in the folder and files name.

asierrl, 2015-12-13:
Thanks a lot, lupus78...
I will try, as soon as I have some 7 hours to try...and let you know, of course.
Best regards,

asierrl, 2015-12-19:
Hi,
That seems to have solved the problem...although a new one appeared. I will post it in a new thread, for more clarity.
Thanks for your help.
 Asier

closed 2016-10-12 by asierrl