OpenDroneMap/ODM #224: ODM stopped at opensfm. Keyerror: u '3382', Code 256

asierrl, 2015-12-19:
I was trying to run OpenDroneMap with 164 images of my own, but the script quit, at opensfm, with Keyerror: u '3382' and code 256.
I have a bunch of log files, but I do not know even what to look for. Any hint?
Below I provide the last lines of the terminarl:
Matching _1060497.jpg  -  84 / 120
2015-12-19 11:46:31,445 reading features
2015-12-19 11:46:32,361 Merging features onto tracks
2015-12-19 11:46:32,368 Good tracks: 527
images 5
nonfisheye images 5
0 partial reconstructions in total.
b
running PYTHONPATH=/home/asier/OpenDroneMap/lib/python2.7/dist-packages "/home/asier/OpenDroneMap/src/OpenSfM/bin/export_bundler" opensfm
Traceback (most recent call last):
  File "/home/asier/OpenDroneMap/src/OpenSfM/bin/export_bundler", line 35, in <module>
    io.export_bundler(data.images(), reconstructions, track_graph, bundle_file_path, list_file_path)
  File "/home/asier/OpenDroneMap/src/OpenSfM/opensfm/io.py", line 57, in export_bundler
    view_list = track_graph[point_id]
  File "/usr/local/lib/python2.7/dist-packages/networkx/classes/graph.py", line 406, in __getitem__
    return self.adj[n]
KeyError: u'3382'
b

quitting cause: 
	PYTHONPATH=/home/asier/OpenDroneMap/lib/python2.7/dist-packages "/home/asier/OpenDroneMap/src/OpenSfM/bin/export_bundler" opensfm
returned with code 256.

Thanks a lot for your suggestions.
Asier

paulinus, 2015-12-19:
It seems that you got very little matches and the reconstruction failed. Do you have the images available somewhere so i can take a look?

asierrl, 2015-12-19:
Thanks, Paulinus. I am now uploading them. They should be available in about two hours (slow uploading conection :-) ) at:

enebada.eu/documentos_publicos/Aldea-ODM.zip 

Best regards, 
 Asier

dakotabenjamin, 2015-12-21:
Gorgeous imagery... I'll see if I can help too.

paulinus, 2015-12-21:
Running opensfm directly on the images worked fine and got the 120 images reconstructed.
[image]
But when running from `run.py`, it gets much fewer matches and only reconstructs 38 images.  There's something is broken in the way we call opensfm, but can see what. I need to investigate more.

igorti, 2015-12-21:
Could it be that quality is insufficient due to resizing that is run prior to opensfm?

dakotabenjamin, 2015-12-21:
Good question. In my experience, matching problems become more prevalent when resized to ~1500px or less. 

@asierrl what parameters did you use with run.py?

paulinus, 2015-12-21:
The resize can affect the results, but in this case i've resized to 1024px.  I tried both pre-resizing as we do in ODM, and telling OpenSfM to resize internally. Both worked. I suspect the problem is something else.

dakotabenjamin, 2015-12-21:
I'm seeing the same thing as @paulinus but I am not able to reproduce @asierrl problem. I am also producing a partial point cloud.

asierrl, 2015-12-21:
Thanks a lot to all of you for your help. I am new to OpenDroneMap, so that, I was just following the wiki and run this command:

 ../OpenDroneMap/run.py --odm_georeferencing-useGcp

I did not say anything before on using a GCP file because I think the problem has nothing to do with it. Could the problem be there?

paulinus, 2015-12-21:
I've found the problem. The default value of `matcher-neighbors` has changed recently from 0 to 8 (https://github.com/OpenDroneMap/OpenDroneMap/blob/gh-pages/run.py#L124).

OpenSfM treats 0 as "match all pairs", which works for sequences without GPS information. 
This sequence has no GPS information, so using `matcher-neighbors=8` yields choosing 8 random neighbors per image.

I'll update OpenSfM to handle the case where it is told to match only X neighbors but there is no GPS info to choose them. Until then, maybe letting the default value to 0 might be safer.

**tl;dr** use
```python /vagrant/OpenDroneMap/run.py --resize-to 1024 --matcher-neighbors 0```

paulinus, 2015-12-21:
I've updated opensfm to ignore the number of neighbors if any image has no GPS info (https://github.com/mapillary/OpenSfM/commit/a0671c43fc004f0a82f6b2c1805ee6a39dd3c109). So `matcher-neighbors` will be ignored for sequences with no GPS.

smathermather, 2015-12-22:
Seems like the default pattern is matcher-neighbors as 0 unless there's GPS ephemeris... .

asierrl, 2015-12-26:
I have tried again and the problem seems to be disappeared now, as it finishes the SfM part of the process. I now get a new error, but being it a different one I will post in a new issue.
Thanks for the help!!
      Asier

closed 2016-10-12 by dakotabenjamin