OpenDroneMap/ODM #225: odm_orthophoto failed to allocate memory; seems to demand too much memory

asierrl, 2015-12-26:
HI, again. 
Once the problem with the SfM matching process has been solved I run into a new one. ODM quits at the orthophoto generating step, due to a lack of memory. I am running ODM on an Ubuntu 14.04 PC, with 32 Gb memory and I7 processing units (with 8 virtual cores). The pictures are in brand new SSD where the only content is this picture folder (164 jpg images), and the ODM folder is another SSD with 17 GB free space. 
The odd thing is that odm_orthphoto seems to demand an absurd amount of memory, as it says it failed to allocate more than 800.000 Gb!! 
What could be the problem? Any checking I can try?
Here you have the whole error reported:

  - running orthophoto generation - Thu Dec 24 12:20:12  2015
running "/home/asier/OpenDroneMap/bin/odm_orthophoto" -inputFile /media/asier/datSSD/Aldea-ODM/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj -logFile /media/asier/datSSD/Aldea-ODM/reconstruction-with-image-size-2400/odm_orthophoto/odm_orthophoto_log.txt -outputFile /media/asier/datSSD/Aldea-ODM/reconstruction-with-image-size-2400-results/odm_orthphoto.png -resolution 20.0 -outputCornerFile /media/asier/datSSD/Aldea-ODM/reconstruction-with-image-size-2400/odm_orthphoto_corners.txt
OpenCV Error: Insufficient memory (Failed to allocate 869698691167844 bytes) in OutOfMemoryError, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/alloc.cpp, line 52
Error in OdmOrthoPhoto:
/build/buildd/opencv-2.4.8+dfsg1/modules/core/src/alloc.cpp:52: error: (-4) Failed to allocate 869698691167844 bytes in function OutOfMemoryError

b

quitting cause: 
	"/home/asier/OpenDroneMap/bin/odm_orthophoto" -inputFile /media/asier/datSSD/Aldea-ODM/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj -logFile /media/asier/datSSD/Aldea-ODM/reconstruction-with-image-size-2400/odm_orthophoto/odm_orthophoto_log.txt -outputFile /media/asier/datSSD/Aldea-ODM/reconstruction-with-image-size-2400-results/odm_orthphoto.png -resolution 20.0 -outputCornerFile /media/asier/datSSD/Aldea-ODM/reconstruction-with-image-size-2400/odm_orthphoto_corners.txt
returned with code 256.

Thanks again.
Best regards, 
       Asier

lupus78, 2015-12-27:
The orthophoto you are trying to create would be too large... you see the error message:
`Failed to allocate 869698691167844 bytes in function OutOfMemoryError`
That's the RAM it tries to allocate.... that's 869698.69GB of RAM :)

You can do two things:
- use the `--odm_orthophoto-resolution` argument, and supply something lower (it is 20pixel/meter by default)
- Usually the Mesh is larger than the actual area you need in you Orthophoto, so what I do is, first run until odm_meshing task, than trim the odm_mesh-0000.ply with Meshlab to the are where it really reconstructed the terrain... and after continue with odm_texturing and orthophoto.

asierrl, 2015-12-28:
Sure, it seems too large, but I thought there might be an issue with estimating the needed RAM. I say that because I've been playing around with different software and I had no problems doing the Orthophoto with them. 
I don't know how to run OpenDroneMap until a given point or how to use the different step-by-step commands. Where can I find some info on them?
Thanks!!
     Asier

lupus78, 2015-12-28:
There is no issue with the calculation, I've checked that.
The default 20px/meter resolution is quite high, thus producing large photos. Open up `odm_orthophoto_log.txt` and you will see what image dimension it tried to create.

asierrl, 2015-12-28:
Thanks lupus78, 
I checked the log file and I found an error, in fact the estimated area of the orthophoto is 543562000000 square meters. The real area should be something around 156250 square meters.
I guess the problem is in some error with my GCP_list. I'll check it and let you know. 
Perhaps, I can first try without the GCPs.
Just to be sure: I am using EPSG: 25829 (ETRS89, UTM 29 North). Is it OK? I read there were some problems with WGS84 UTM 17N, but I understood (perhaps misunderstood) that they had been solved. 
Thanks a lot, again.
Asier

lupus78, 2015-12-28:
The GCP format description is here in the 2nd paragraph:
https://github.com/OpenDroneMap/OpenDroneMap/wiki/2.-Running-OpenDroneMap

However if you have an older build, you may still have a buggos code which expects the pixel coordinates in reserve order, see #213 
You can check the `Mean georeference error` in `odm_georeferencing_log.txt`

asierrl, 2015-12-28:
Thank you, lupus78. I found the error. A number was missing from one of the points.
But I also found that many of the points were disregarded with a message saying that the corresponding image was not found. Those images were there. I introduced each control point many times, for each of the images where it is visible. I now have seen that only the best three are used, so that I would not need such a redundant GCP file. But, could this be the reason for disregarding so many GCPs (based on the imposibility to find the images)?
Best regards,
      Asier

lupus78, 2015-12-28:
nope, georeferencing is only using the 3 best fitting GCP, no matter how much you specify.

closed 2016-10-12 by asierrl