OpenDroneMap/ODM #226: How to upload an updated ccd_defs.json?

asierrl, 2016-01-07:
I updated my ccd_defs.json file and I would like to upload it; my ridiculously small contribution. I was following the instructions in the README.md file, which should be 1) check if it compiles and 2) do a pull request.
I don't even fully understand what "check it it compiles" mean, but I entered ccd_defs_check.py in a terminal and I got a "command not found" message.
What am I doing wrong?
Thanks!!
Asier

dakotabenjamin, 2016-01-07:
try `python ccd_defs_check.py`

asierrl, 2016-01-08:
Thanks. It works!! Although, only from the OpenDroneMap folder.

dakotabenjamin, 2016-01-08:
That's right. I look forward to the pull request :+1:

asierrl, 2016-01-11:
I tried...but I am afraid I didn't understand all the process. I'm knew to GitHub and some of the concepts are not clear to me...I have to come back on the issue again, and see if I can understand it all in a second read...
Thanks for your support.

dakotabenjamin, 2016-01-11:
No worries! Here's some help that I found when I was learning git: 

https://rogerdudler.github.io/git-guide/
http://git-scm.com/book/en/v2/Getting-Started-Git-Basics
https://try.github.io/levels/1/challenges/1

asierrl, 2016-01-13:
Thanks!! Those were really useful...after a couple of hours of reading I've been able to complete muy pull request.

dakotabenjamin, 2016-01-13:
Nice! :+1:

closed 2016-01-13 by dakotabenjamin