OpenDroneMap/ODM #227: csfm.so: undefined symbol

squeakus, 2016-01-10:
Hi, Apologies if this is an opensfm issue rather than an opendronemaps issue. I compiled the default branch of opendrone maps (gh-pages) which uses cmake. I hit issue #178 when building as cmake couldn't find my opencv libraries, despite the default ubuntu libraries being installed. I managed to fix it by manually editing the link.txt files and adding the libraries.

Now when I execute run.py I get the following error:
ImportError: /home/jonathan/OpenDroneMap/src/OpenSfM/opensfm/csfm.so:
undefined symbol: _ZN2cv14findHomographyERKNS_11_InputArrayES2_idRKNS_12_OutputArrayEid

A similar error was reported in issue #197 and it looks like moving the build system to cmake should have fixed it( #201), is it that I am linking to the wrong opencv version, what version should I be linking to?

squeakus, 2016-01-10:
After a bit more digging it turns out the openCV problem has to do with picking up version 3 and has been fixed already (#210). I managed to get it to build perfectly using the docker container and still I get the same error from openSFM. I rebuilt openSFM using: python setup.py build and there were no errors. The error seems to be specific to the csfm.so file.

paulinus, 2016-01-11:
I don't know what's happening here, but I would try deleting `opensfm/build` and `opensfm/cmake_build` before rebuilding using `python setup.py build` to make sure that anything that you've changed outside opensfm gets properly linked.

squeakus, 2016-01-11:
When I delete the build and cmake_build folders and rerun cmake I get the following error:
-- Could NOT find ceres (missing:  CERES_LIBRARY CERES_INCLUDE_DIR) 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:

I grepped the folder but I couldn't find where these variables are set. I ran ccmake and there was no option to set them and I manually set them but I still got the same error. It is successfully downloading and building the ceres solver. Do you know where these variables are set?

dakotabenjamin, 2016-01-13:
Just to be sure, what's your OS?

squeakus, 2016-01-13:
Ubuntu 14.04. I managed to get my hands on an old machine, do a clean OS install and opendronemaps worked so it looks like it is some weirdness specific to my current machine.

dakotabenjamin, 2016-01-13:
Alright, that's good to know. The changes we are working on over at https://github.com/edgarriba/OpenDroneMap should fix the contamination issues in the future.

closed 2016-10-12 by dakotabenjamin