OpenDroneMap/ODM #228: ODM stopping at openSFM if images resized larger than 1000

darkpioneer, 2016-01-11:
Hi,
I am getting a strange issue similar to https://github.com/OpenDroneMap/OpenDroneMap/issues/224

OpenSFM only seems to break if i resize the my images to anything above 1000
If i specify "run.py -resize-to 1000" or less it runs fine.
They where taken with a phantom 3 pro useing the Pix4D Capture app on android.

here is a link to the images https://www.dropbox.com/s/0mg57j1qcwlekfg/mission1.tar.gz?dl=0

```
2016-01-11 01:33:00,262 reading features
2016-01-11 01:33:00,440 Merging features onto tracks
2016-01-11 01:33:00,979 Good tracks: 22257
Images 15
nonfisheye images 15
0 partial reconstructions in total.
Traceback (most recent call last):
  File "/mnt/OpenDroneMap/src/OpenSfM/bin/mesh", line 23, in <module>
    reconstructions = data.load_reconstruction()
  File "/mnt/OpenDroneMap/src/OpenSfM/opensfm/dataset.py", line 291, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: 'opensfm/reconstruction.json'
b

quitting cause: 
	PYTHONPATH=/mnt/OpenDroneMap/lib/python2.7/dist-packages "/mnt/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.
```

paulinus, 2016-01-11:
OpenSfM is not reconstructing the scene because it finds it to be too flat. I've opened an issue for this here https://github.com/mapillary/OpenSfM/issues/61
In the meantime, the following patch works for your dataset, by just skipping the flatness check.

```
diff --git a/opensfm/reconstruction.py b/opensfm/reconstruction.py
index 3403bfd..f14d20b 100644
--- a/opensfm/reconstruction.py
+++ b/opensfm/reconstruction.py
@@ -143,7 +143,7 @@ def bundle_single_view(graph, reconstruction, shot_id, config):
 def pairwise_reconstructability(common_tracks, homography_inliers):
     outliers = common_tracks - homography_inliers
     outlier_ratio = float(outliers) / common_tracks
-    if outlier_ratio > 0.3:
+    if outlier_ratio > 0.0:
         return common_tracks
     else:
         return 0
```

dakotabenjamin, 2016-01-11:
thanks @paulinus, I was able to replicate the issue but didn't understand why.

darkpioneer, 2016-01-11:
Thanks @paulinus it was quite confusing, I will try the fix.
This was the 1st time I tried with my own data, going to fly again at the farm but maybe over some trees and barns as well.

closed 2016-01-11 by darkpioneer