OpenDroneMap/ODM #231: gaps in my point cloud

asierrl, 2016-01-16:
I got some notorious gaps in my point cloud and I was thinking on increasing the image size for image matching. However, if I am not wrong, the stated image size (I previously worked with the default of 2400) only affects to OpenSfm and PMVS, am I? Hence, I thought that maybe by increasing the image size I could get more match points in those areas. 
Or should I try with a smaller pmvs-csize?
By the way, is there a way to tell OpenDroneMap not to reduce the image size? I mean just use the whole images? It is easy to ask for a resizing size equal to the actual size when all the pictures are the same size, but I was thinking in cases where image sizes differ?
Thanks a lot.
       Asier

Fi156, 2016-01-16:
To my understanding:
Increasing the image size can decrease the output quality. No image feature looks exactly the same in both images, the more pixel the algorithm get, the bigger will be the difference in possibly matching features. 

If I where you, I would tweak the following parameters:
--min-num-features
--pmvs-csize
--pmvs-threshold

smathermather, 2016-01-18:
@Fi156 @asierrl All good advice. Yes, if you want a denser point cloud, sometimes high resolution images help, but only if you are already getting matches. If there are big gaps without any matches, resampling to lower resolution might help.

asierrl, 2016-01-19:
Thanks a lot. By the moment I will try reducing the image size, and later I will increase min-num-features and pmvs-csize. I am not sure about pmvs-threshold...I guess I should decrease it, shouldn't I?

asierrl, 2016-01-21:
Up to now I followed smathermather's advise, and run OpenDroneMap with 1200 picture size. It's true that the gap I had is now much more populated. But it has been atthe cost of loosing some large regions, where the overlap among pictures was smaller (due to strong winds during the flight). 
I have a couple of images to show this, in case you are interested.

dakotabenjamin, 2016-01-21:
One solution you can try is to increase min-num-features which would likely shift more points into a single bundle. This will probably catch the other regions, since OpenSfM only reconstructs the largest bundle.

Fi156, 2016-01-23:
pmvs-threshold: Yes, decreasing should be the way to go, if you want to get a denser reconstruction from poor datasets.

closed 2016-10-12 by dakotabenjamin