OpenDroneMap/ODM #235: orthophoto generation before texturing?

rionlerm, 2016-01-25:
Hi guys.  I must say I am quite impressed the way odm runs on Ubuntu 12 with datasets of >200 photos from our DJI P3A.

However, would it be possible to modify the script to generate orthophoto first and NOT from texturing? The current or default generation creates a photo where the trees look 3D from above and not what it would look like from UAV.

smathermather, 2016-01-25:
Hi @rionlerm, excellent question. What follows is a long explanation; TLDR: it's something we are interested in adding.

An argument could be made for doing texturing based on a DEM, instead of the DSM. This is how photogrammetry generally does this, and in the process stitch lines are chosen in order to minimize the affect of the different perspectives. There are both drawbacks and advantages to each approach, but having the flexibility of choice in ODM would be great.

To build this functionality into ODM, we'd need to integrate a point cloud classifier. I think this could be done directly in PCL, or with the integration of PDAL (http://www.pdal.io/tutorial/pcl_ground.html#pcl-ground), but I haven't looked deeply into it yet. (I have used the tool in PDAL, but the build process to get there is painful ATM... .)

Then, we build a mesh from the ground classified points, create the mesh, texture the mesh, and then generate the orthophoto. There was some conversation on gitter https://gitter.im/OpenDroneMap/OpenDroneMap about this around December 14th. @GeospatialDaryl thought it better to re-implement directly in PCL, rather than pull in all the PDAL dependencies. Now that I've built PDAL a few times, I lean toward this approach.

eyeNsky, 2016-03-01:
In the past we have used -thin_with_grid to make a quick "bare earth" model from a lidar point cloud to orthorectify imagery. By using a 5-20 m grid you end up with a relatively flat surface for the textured mesh. Won't be perfect in dense vegetation, but would reduce the 3D look of the final ortho.

eyeNsky, 2016-03-01:
sorry, lasthin was the tool we used.

smathermather, 2016-03-01:
Could also use this: http://www.pdal.io/stages/filters.ground.html#filters-ground now that pdal is integrated. It would take a little more effort on the PDAL to ensure that all the bits and bobs were in place, but not much effort, I'd hazard. But, it wouldn't save time in processing, just change the overall result.

dakotabenjamin, 2016-03-01:
Note: PDAL is integrated into the edgarriba-cmake branch, not the main branch *yet*.

rionlerm, 2016-03-02:
@dakotabenjamin can I make use of the edgarriba-cmake branch and what runtime paras' should I include? I would like to experiment with this feature on our datasets.

dakotabenjamin, 2016-03-02:
In the project base directory:
```
git checkout edgarriba-cmake
export PYTHONPATH=$PYTHONPATH:`pwd`/SuperBuild/install/lib/python2.7/dist-packages
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:`pwd`/SuperBuild/install/lib
bash configure.sh
mkdir build && cd build && cmake .. && make && cd ..
```

smathermather, 2016-03-03:
(Also, we'll be removing the ```export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:`pwd`/SuperBuild/install/lib``` bad habit in the near future before merging with the main branch.)

bryanibit, 2017-04-13:
@rionlerm Did you make it? Does it accelerate the ODM for orthophoto by removing texturing cell?

rionlerm, 2017-04-19:
@bryanibit I have run some data sets since this post in 2016 and in hindsight it seems faster however, seam levelling seems to be a current issue for me in master branch hence, I use a work-around branch that Stephen Mather produced (cannot find the repository these days). Also the environmental variables above still need to be set in ODM as far as I know?

bryanibit, 2017-04-19:
@rionlerm Now the module of orthophoto needs texturing mesh as input file. I wonder what you use as input to produce orthophoto.

rionlerm, 2017-04-19:
@dakotabenjamin see above?

dakotabenjamin, 2017-04-20:
Yeah @bryanibit is correct you can't skip texturing. 

We merged the seam-leveling fix in `master`, by the way.

dakotabenjamin, 2017-04-20:
With the #542 pull request, the orthophoto should look more realistic.

bryanibit, 2017-04-20:
@dakotabenjamin Every step is not trivial or removable for producing orthophoto.

smathermather, 2020-04-14:
I would say this is (in spirit) addressed by the introduction of `--fast-orthophoto`. I will close for now.

closed 2020-04-14 by smathermather