OpenDroneMap/ODM #237: ccd_defs is soon to be depreciated

dakotabenjamin, 2016-01-26:
Since we have more or less removed support for matcher/bundler (it still exists in the code, but OpenSfM is faster and better and therefore runs as the default), ccd_defs.json has been replaced with OpenSfM's own [sensor data file](https://github.com/mapillary/OpenSfM/blob/master/opensfm/data/sensor_data.json).

I will not be merging any ccd_defs.json update PRs anymore.

renzoer69, 2016-03-24:
Micasense RedEdge Size CCD?

dakotabenjamin, 2016-03-24:
I don't know how a multispectral camera would work with ODM. I think you'll have to get in contact with Micasense to get more information on their sensors.

smathermather, 2016-03-24:
RedEdge has 5 bands, correct? ODM assumes 3 bands at this time, but we're open to contributions which extend that to multispectral.

renzoer69, 2016-03-24:
Yes, RedEdge has 5 band.
I going to work with one band each time.

dakotabenjamin, 2016-03-24:
That being said, I don't think OpenSfM actually 'fails' when there's no ccd width information.

closed 2016-10-12 by dakotabenjamin