OpenDroneMap/ODM #238: Stopped

piticu, 2016-02-01:
Hi guys. I've let odm run overnight over a large (for me at least, 372 images) and got this error:

`Painting the reconstruction from 372 cameras
Done.
Reconstruction 0 : 372 images , 336863 points
1 partial reconstructions in total.
Traceback (most recent call last):
  File "/odm_app/OpenDroneMap/src/OpenSfM/bin/mesh", line 32, in <module>
    data.save_reconstruction(reconstructions, filename='reconstruction.meshed.js
on')
  File "/odm_app/OpenDroneMap/src/OpenSfM/opensfm/dataset.py", line 297, in save
_reconstruction
    fout.write(io.json_dumps(reconstruction))
  File "/odm_app/OpenDroneMap/src/OpenSfM/opensfm/io.py", line 209, in json_dump
s
    return json.dumps(data, indent=indent, ensure_ascii=False).encode(codec)
  File "/usr/lib/python2.7/json/__init__.py", line 250, in dumps
    sort_keys=sort_keys, **kw).encode(obj)
  File "/usr/lib/python2.7/json/encoder.py", line 210, in encode
    return ''.join(chunks)
MemoryError
b

quitting cause:
        PYTHONPATH=/odm_app/OpenDroneMap/lib/python2.7/dist-packages "/odm_app/O
penDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.`

Most probably hard-drive got into sleep and the virtual machine didn't noticed. Is there any posibility to resume work where it was interrupted so i don't have to start it all over again?

paulinus, 2016-02-01:
It looks like opensfm's `bin/mesh` run out of memory (ram memory).

Mesh is not required for the rest of the OpenDroneMap pipeline. You can try editing `src/opensfm/bin/run_all` and removing the last line.

(We should modify the way OpenDroneMap calls OpenSfM so that we don't call mesh)

rionlerm, 2016-02-01:
I found the script stops at

Expand patches...

@smathermather or @lupus78   is this still memory issues on my Ubuntu 4G RAM laptop?
Used 203 images

dakotabenjamin, 2016-02-10:
@rionlerm probably. It would probably be fine if you were dedicating all 4GB to the process but I doubt that is the case.

rionlerm, 2016-02-15:
How do I dedicate all 4GB?

dakotabenjamin, 2016-02-16:
It's impossible, since the operating system needs at least some RAM to operate, but you could try to minimize other non-essential processes. To be honest I don't know how much it would help, though.

rionlerm, 2016-02-17:
Thanks we can close this!

closed 2016-02-17 by smathermather