OpenDroneMap/ODM #239: Error message when running georef

bobit200, 2016-02-02:
Hi,
Just wondering if anyone can help with the error message attached.  Not sure if it is a bug or an issue with my gcp file (also attached).  Open Drone Map creates my textured mesh of the runway at LBA fine (100 images) and I have created the gcp file with UTM ref points in order to produce the Orthophoto.

Thanks
[image]
[Uploading gcp_list.txt…]()

bobit200, 2016-02-02:
[image]

bobit200, 2016-02-02:
WGS84 UTM 30U
587326.86 5970228.50 0 2201 605 img_4976.jpg
587339.60 5970215.16 0 2165 821 img_4981.jpg
587339.60 5970215.16 0 2029 1973 img_4985.jpg
587356.51 5970197.23 0 2209 333 img_4985.jpg
587377.98 5970174.37 0 1865 745 img_4992.jpg
587395.16 5970156.22 0 1921 3041 img_5000.jpg
587442.65 5970106.59 0 2297 1305 img_5006.jpg
587463.20 5970084.27 0 2173 2309 img_5011.jpg
587503.46 5970045.18 0 2169 2533 img_5017.jpg
587536.14 5970003.39 0 2453 1869 img_5021.jpg
587587.85 5969953.62 0 2265 1577 img_5027.jpg
587629.00 5969909.43 0 2473 1813 img_5033.jpg
587691.20 5969843.72 0 2013 1153 img_5039.jpg
587732.22 5969799.75 0 2417 1141 img_5045.jpg
587752.15 5969778.65 0 2321 2817 img_5051.jpg
587815.16 5969712.28 0 2349 1721 img_5060.jpg
587855.97 5969668.53 0 2337 1637 img_5067.jpg
587897.33 5969624.24 0 2469 637 img_5072.jpg
587917.60 5969602.25 0 2477 941 img_5076.jpg

dakotabenjamin, 2016-02-02:
I think that instead of UTM 30U you should have UTM 30N

bobit200, 2016-02-02:
Thanks will try later.  Why Zone 30N when I am in 30U?

dakotabenjamin, 2016-02-02:
The N stands for Northern hemisphere, the program doesn't use the latitude
bands because the easting and northing with the hemisphere info is
sufficient.

It's confusing, but that's why we added the feature to use EPSG codes
instead.

bobit200, 2016-02-03:
changed to 30N and it ran fine.  Another question.  I used the coal oil point example to work out how to use odm.  Last night I imported the generated geoTiff into Google Maps Pro and it placed the image in the ocean! When generating the gcp file for the example I converted the lat long coordinates into UTM using the www.latlong.net website. Are EPSG codes more reliable than UTM? or does Google only work with lat long? Steep learning curve al this!

yjmenezes, 2016-02-03:
@bobit200 

EPSG  codes carry a lot of information. http://spatialreference.org/ref/epsg/
Google use WebMercator https://en.wikipedia.org/wiki/Web_Mercator,  EPSG:3857
You must reproject from UTM to WebMercator.

Maybe the first CGP line could have a space between Zone and Hemisphere.
WGS84 UTM 30 N 
I agree with @dakotabenjamin , EPSG codes are better.

[]s
julio menezes

dakotabenjamin, 2016-02-03:
Re EPSG codes: the way the code parses that line, using EPSG codes is far less prone to human error than UTM. 

Personally I'm in favor of gutting UTM completely, because of the confusion @bobit200 had.

bobit200, 2016-02-03:
Thanks for the info.  Any pointers on how I can convert lat/long or UTM to EPSG:3857?

dakotabenjamin, 2016-02-03:
http://cs2cs.mygeodata.eu/

bobit200, 2016-02-03:
thanks for your input. I will try and get my head around these with the coal example.

closed 2016-02-03 by bobit200