OpenDroneMap/ODM #240: S_create_c_locale name not valid

shomy4, 2016-02-02:
I have a problem, i tried to stitch images but it failed and inside odm_georeferencing_log i got this error

`Missing input in order to use GCP for georeferencing. Using EXIF data instead.

Geographical reference system
WGS84 UTM 34N
514037 4982275

Choosing optimal camera triplet...
Mean georeference error 54.656
... optimal camera triplet chosen:
20, 21, 28

Final transform:
[ 0.74487984,	-0.39607588,	0.44562403,	85.53311710 ]
[ -0.38514911,	0.22468194,	0.84349336,	-32.36983278 ]
[ -0.45510113,	-0.83841802,	0.01552542,	195.53457025 ]
[ 0.00000000,	0.00000000,	0.00000000,	1.00000000 ]

Reading mesh file...
Error in Georef:
locale::facet::_S_create_c_locale name not valid`

What could be the cause of this error?

burlicconi, 2016-02-03:
I have similar problem, seeking for answer but no luck for now.

Hope that good people can help us here :)

dakotabenjamin, 2016-02-03:
Try:
```
export LC_ALL=C
```

and then reinstall

shomy4, 2016-02-03:
That did it.
Thanks!

closed 2016-02-03 by shomy4