OpenDroneMap/ODM #241: vagrant up ... chown fails.

mrosen, 2016-02-03:
I'm unable to get the vagrant machine set up properly:  "chown: changing ownership of ‘/vagrant’: Not a directory".  I'm stopped cold right here.  Any thoughts?

Here's what happened when I followed the instruction for get up on odm/vagrant (https://github.com/OpenDroneMap/odm_vagrant)

C:\SFM_Learning\odm\odm_vagrant [master]> vagrant up
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Importing base box 'ubuntu/trusty64'...
==> default: Matching MAC address for NAT networking...
==> default: Checking if box 'ubuntu/trusty64' is up to date...
==> default: Setting the name of the VM: odm_vagrant_default_1454531118355_95692
==> default: Clearing any previously set forwarded ports...
==> default: Fixed port collision for 22 => 2222. Now on port 2200.
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
==> default: Forwarding ports...
    default: 22 (guest) => 2200 (host) (adapter 1)
==> default: Running 'pre-boot' VM customizations...
==> default: Booting VM...
==> default: Waiting for machine to boot. This may take a few minutes...
    default: SSH address: 127.0.0.1:2200
    default: SSH username: vagrant
    default: SSH auth method: private key
    default:
    default: Vagrant insecure key detected. Vagrant will automatically replace
    default: this with a newly generated keypair for better security.
    default:
    default: Inserting generated public key within guest...
    default: Removing insecure key from the guest if it's present...
    default: Key inserted! Disconnecting and reconnecting using new SSH key...
==> default: Machine booted and ready!
==> default: Checking for guest additions in VM...
    default: The guest additions on this VM do not match the installed version of
    default: VirtualBox! In most cases this is fine, but in rare cases it can
    default: prevent things such as shared folders from working properly. If you see
    default: shared folder errors, please make sure the guest additions within the
    default: virtual machine match the version of VirtualBox you have installed on
    default: your host and reload your VM.
    default:
    default: Guest Additions Version: 4.3.36
    default: VirtualBox Version: 5.0
==> default: Mounting shared folders...
    default: /vagrant => C:/SFM_Learning/odm/odm_vagrant
The following SSH command responded with a non-zero exit status.
Vagrant assumes that this means the command failed!

chown `id -u vagrant`:`id -g vagrant` /vagrant

Stdout from the command:

Stderr from the command:

stdin: is not a tty
chown: changing ownership of ‘/vagrant’: Not a directory

dakotabenjamin, 2016-02-03:
What version of Vagrant are you running? 

I don't think this is a ODM problem: https://github.com/mitchellh/vagrant/issues/5933

mrosen, 2016-02-04:
Thanks for the reply.  I confirm that this is a Vagrant problem.  Updating from 5.0.4 to 5.0.14 fixed it.  Moving on...

closed 2016-02-04 by dakotabenjamin