OpenDroneMap/ODM #242: pacifica dataset fails ? "IOError: [Errno 21] Is a directory: 'opensfm/'"

mrosen, 2016-02-04:
Working through instructions for running under Vagrant (https://github.com/OpenDroneMap/odm_vagrant).

Step 12:  "Run the OpenDroneMap App on the an odm_data test dataset." fails as seen below.  What's wrong?

====================

vagrant@vagrant-ubuntu-trusty-64:/vagrant_data/odm_data/pacifica$ /odm_app/OpenDroneMap/run.py

  - configuration:
     start_with : resize
     end_with : odm_orthophoto
     force_focal : None
     odm_texturing_textureWithSize : 3600
     matcher_neighbors : 8
     matcher_threshold : 2.0
     cmvs_maxImages : 500
     odm_orthophoto_resolution : 20.0
     pmvs_threshold : 0.7
     zip_results : False
     resize_to : 2400
     use_opensfm : True
     min_num_features : 6000
     odm_meshing_octreeDepth : 9
     odm_meshing_solverDivide : 9
     pmvs_wsize : 7
     odm_texturing_textureResolution : 4096
     matcher_ratio : 0.6
     odm_meshing_samplesPerNode : 1.0
     odm_meshing_maxVertexCount : 100000
     matcher_distance : 0
     pmvs_level : 1
     run_only : None
     job_dir_name : None
     odm_georeferencing_useGcp : False
     odm_georeferencing_gcpFile : gcp_list.txt
     pmvs_csize : 2
     pmvs_minImageNum : 3
     force_ccd : None

  - source files - Thu Feb 04 18:19:54  2016

    found 0 usable images

  using max image size of 2400 x 2400

  - preparing images - Thu Feb 04 18:19:54  2016

  - running OpenSfM - Thu Feb 04 18:19:54  2016
running import_bundler
running PYTHONPATH=/odm_app/OpenDroneMap/lib/python2.7/dist-packages "/odm_app/OpenDroneMap/src/OpenSfM/bin/import_bundler" opensfm --list list.txt
No bundle.out given.  Importing only the list of images list.txt.
Converting output from Bundler to OpenSfM
b
running PYTHONPATH=/odm_app/OpenDroneMap/lib/python2.7/dist-packages "/odm_app/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
2016-02-04 18:19:55,607 Extracting focal lengths for image
Traceback (most recent call last):
  File "/odm_app/OpenDroneMap/src/OpenSfM/bin/focal_from_exif", line 27, in <module>
    d = exif.extract_exif_from_file(data.load_image(image))
  File "/odm_app/OpenDroneMap/src/OpenSfM/opensfm/dataset.py", line 60, in load_image
    return open(self.__image_file(image))
IOError: [Errno 21] Is a directory: 'opensfm/'
b

quitting cause:
        PYTHONPATH=/odm_app/OpenDroneMap/lib/python2.7/dist-packages "/odm_app/OpenDroneMap/src/OpenSfM/bin/run_all" opensfm
returned with code 256.

vagrant@vagrant-ubuntu-trusty-64:/vagrant_data/odm_data/pacifica$

dakotabenjamin, 2016-02-04:
This issue looks similar to: https://github.com/OpenDroneMap/OpenDroneMap/issues/224

Try running: `run.py --resize-to 1024 --matcher-neighbors 0`

dakotabenjamin, 2016-02-04:
Also: caliterra, seneca, and [toledo](https://github.com/opendronemap/odm_data_toledo) are datasets that are generally successful tests if you're trying to get the hang of ODM.

mrosen, 2016-02-05:
*groan*.  You'll love this one.  Step 9 from https://github.com/OpenDroneMap/odm_vagrant:

> Clone the sample imagery repository https://github.com/OpenDroneMap/odm_data.git
> 
> sudo apt-get -y install git
> cd /vagrant_data/
> git clone https://github.com/OpenDroneMap/odm_data.git

Sadly, results in a largely empty directory structure.  To get the actual data you need to follow the instruction from the Readme file:
>git clone --recursive git://github.com/OpenDroneMap/odm_data

That is, the reason it was failing was because there was no input data!  A fix here would be to update Step 9.

dakotabenjamin, 2016-02-08:
Fixed!

closed 2016-03-08 by dakotabenjamin