OpenDroneMap/ODM #243: returned with code 35584

bobit200, 2016-02-08:
Hi,
Any had any issues with this error code.  Trying to produce an orphophoto but keep getting this message on completion.  My mesh is created fine and I am using a Gcp file for my reference points.
[image]
[image]

bobit200, 2016-02-08:
Anyone got any ideas?

dakotabenjamin, 2016-02-09:
What do you get from `reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt`?

bobit200, 2016-02-10:
hi,
that file is empty - 0 bytes.  It does exist.

Tried running it again but with different UTM reference points and increased the number to ten. Three of the points in the original gcp file were the same UTM point but on different images. odm now produced an orthophoto with the new gcp file.

Thanks.

dakotabenjamin, 2016-02-10:
Perfect. Is the problem solved?

bobit200, 2016-02-11:
Yes.  Issue closed.

closed 2016-02-11 by bobit200