OpenDroneMap/ODM #244: boruszyn_kap dataset fails, "Translating source or target SRS failed" using "PUWG1992"

mrosen, 2016-02-08:
I'm trying to get ODM to work with the boruszyn dataset using the supplied gcp_list.txt file.  But it fails.  The gcp_list.txt references "PUWG1992." which seems unrecognized.  I got a similar error trying to run the COPR dataset ("WGS84 longlat" seem unrecognized).  

How can I make the GCP list work?

Here's what I ran:
vagrant@vagrant-ubuntu-trusty-64:/vagrant_data/odm_data/boruszyn_kap$ /odm_app/OpenDroneMap/run.py  --odm_georeferencing-gcpFile gcp_list.txt --odm_georeferencing-useGcp

Here's how it failed:
...

  - running georeferencing - Mon Feb 08 20:14:54  2016
running "/odm_app/OpenDroneMap/bin/odm_georef" -bundleFile /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400/pmvs/bundle.rd.out -gcpFile /vagrant_data/odm_data/boruszyn_kap/gcp_list.txt -imagesPath /vagrant_data/odm_data/boruszyn_kap/ -imagesListPath /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400/pmvs/list.rd.txt -bundleResizedTo 2400 -inputFile /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model.obj -outputFile /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo.obj -outputCoordFile /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/coordFile.txt -inputPointCloudFile /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000.ply -outputPointCloudFile /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400-results/option-0000_georef.ply -logFile /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /vagrant_data/odm_data/boruszyn_kap/reconstruction-with-image-size-2400-results/odm_texturing/odm_textured_model_geo_georef_system.txt
b
ERROR 1: Translating source or target SRS failed:
PUWG1992
Traceback (most recent call last):
  File "/odm_app/OpenDroneMap/run.py", line 999, in <module>
    func()
  File "/odm_app/OpenDroneMap/run.py", line 889, in odm_georeferencing
    lat_frac = coord_to_fractions(coords_wgs84[1],['N','S'])
IndexError: list index out of range
vagrant@vagrant-ubuntu-trusty-64:/vagrant_data/odm_data/boruszyn_kap$ /odm_app/Open
DroneMap/run.py  --odm_georeferencing-gcpFile gcp_list.txt --odm_georeferencing-use
Gcp

dakotabenjamin, 2016-02-09:
Try using the following: `--odm_georeferencing-gcpFile gcp_list_wgs84.txt`
Also you will need to change the first line of gcp_list_wgs84.txt to EPSG:4326

smathermather, 2016-02-10:
Now if only we knew the correct EPSG for that Polish coordinate system for Boruzyn... @merkato -- do you know the EPSG code for the Boruzyn dataset? BTW, I have a pull request in for changing the first line of the WGS84 version of the GCP file.

merkato, 2016-02-11:
It is EPSG:2180 crs

smathermather, 2016-02-11:
Awesome. Did a pull request. Thanks @merkato.

merkato, 2016-02-12:
Yes, It's awesome. I've recreated model with this gcp file, and results are great. In the center of reconstruction ground error is under 0.2 m. On the edges, error is significant - no gcp's there, bowl effect, rather oblique than nadir photos. But clearly that's awesome. ;)

closed 2016-10-12 by dakotabenjamin