OpenDroneMap/ODM #245: How Can I get an orthophoto if I don't have a geographic reference system

mrosen, 2016-02-10:
Suppose I have have a collection of overlapping drone images but no GPS ephemeris and no GCPs.  I think I can still get through the meshing and texturing steps (right?).  Is there some way I can get odm_orthophoto to produce a rectified image?  Seems like there should be some value of resolution which would allow this.

smathermather, 2016-02-10:
There's no magic way here-- even if you could find an appropriate resolution, you have no way of knowing which is your z axis. Agisoft gets around this by letting you interactively choose a view and choose to create an orthophoto from that view, but that requires a graphical interface.

dakotabenjamin, 2016-02-10:
You can create a GCP file by hand, if you have accurate coordinates of a few (at least 3) landmarks in the images. I've seen instances of GCPs where the z axis was set to 0 or maybe you could guess. Note that it probably won't be as accurate.

mrosen, 2016-02-10:
Can I ask for a little clarification on this:  what do we mean by confusion about the z axis.  We get a properly textured point cloud.  We know where the xy plane is in that reference system.  If we orthographically project down onto that, is that not the orthorectified image we seek?

dakotabenjamin, 2016-02-10:
Well, "we" (humans, not computers) know the where the xy plane is in the textured model because we realize that it should be a planar surface. The software, OTOH, has no idea where x, y, or z is without a geographic reference. The orthorectified image could be projected from the 3D surface from any angle. When you've got a geographic reference, then there's a (0,0,0) point somewhere that represents real space-- the points or pixels or whatever in the surface are able to be oriented relative to that reference, and then the images can be projected "down" onto that surface (because now it has knowledge of what down is).

mrosen, 2016-02-10:
I'm not following Benjamin's explanation.  The point cloud already has a reference system.  Why can't we use that?  When we do a 3D rendering of the point cloud, we can position the camera in coordinates from the cloud's reference system.  As we move the z value out to infinity (and keeping x and y in the middle of their domains), I think the resulting image becomes the orthorectified image we seek.  Right?  We can do that without a UI.  What am I missing?

smathermather, 2016-02-12:
The point cloud has an arbitrary reference system. I could be rotated 180 degrees from normal on any or all axes, so in order to perform what you suggest, we still need to get the Z axis correct. It's true that this does not necessarily require a graphical interface, but I can't see any trivial solution. Dakota is out of town at the moment, but maybe when he gets back he can post some screen shots of the difference between an unreferenced mesh and a referenced one.

liwenzhuo, 2016-04-10:
hi~ Could anyone please tell me how does ODM process tie points after matching, then feed them to bundler? To my knowledge, bundler need a initial 3D coordinates. The question is how to estimate the initial 3D coordinates for tie points.

smathermather, 2016-12-20:
Related: https://github.com/OpenDroneMap/OpenDroneMap/pull/440

dakotabenjamin, 2017-01-19:
With #440 you can now get an ortho, but no guarantees that it will be useful.

pierotofy, 2020-03-19:
For a while ODM has been generating orthophotos even for non-georeferenced datasets. This can be closed. :clinking_glasses:

closed 2020-03-19 by pierotofy

smathermather, 2020-03-19:
Here's an orthophoto of my espresso making setup as deployed in Rwanda:
[image]

It makes a nice surface model too:
[image]

pierotofy, 2020-03-19:
lol