OpenDroneMap/ODM #247: Is it available to run ODM on Windows

xialang2012, 2016-02-18:
Just curious... I have checked the libraries which used by ODM, it seems most of the libraries have the Windows version, except the gsl.

dakotabenjamin, 2016-02-18:
I've never tried to build it on windows, as is I don't think it will work. But if what you say is true, maybe it could be installed manually with [gsl for windows](http://gnuwin32.sourceforge.net/packages/gsl.htm)

xialang2012, 2016-02-19:
Okay, If I have enough time, I will try it.

xialang2012, 2016-02-24:
I have already successful built ODM on Windows with VS 2013.

dakotabenjamin, 2016-02-24:
I will have to try this! What did you do to make it work?

smathermather, 2016-02-24:
Whoa! Awesome.

xialang2012, 2016-02-25:
@dakotabenjamin The  compatibility of the ODM is very well, I have never revised any codes on it. Just built Opencv 3.0, PCL 1.7.2 (it is wise to use all in to installer), PROJ 4.8 on Windows with VS 2013, then built the modules of the ODM respectively. The include and library paths, the libraries used are as below:

> <IncludePath>C:\Program Files\PCL 1.7.2\include\pcl-1.7;C:\Program Files\PCL 1.7.2\3rdParty\Eigen\eigen3;C:\Program Files\PCL 1.7.2\3rdParty\Boost\include\boost-1_55;C:\Program Files\PCL 1.7.2\3rdParty\FLANN\include;C:\Program Files\PCL 1.7.2\3rdParty\QHull\include;C:\Program Files\PCL 1.7.2\3rdParty\VTK\include\vtk-5.10;D:\opencv3\build\include;E:\gdal\proj-4.8.0\src;

> <LibraryPath>C:\Program Files\PCL 1.7.2\lib;C:\Program Files\PCL 1.7.2\3rdParty\Boost\lib;C:\Program Files\PCL 1.7.2\3rdParty\VTK\lib\vtk-5.10;E:\opencv3\lib\Release;E:\gdal\proj-4.8.0\src;

><AdditionalDependencies>pcl_common_release.lib;pcl_io_release.lib;pcl_surface_release.lib;pcl_kdtree_release.lib;vtkCommon.lib;opencv_core300.lib;opencv_imgproc300.lib;opencv_imgcodecs300.lib;opencv_highgui300.lib;proj_i.lib;

ahmadusman, 2016-03-04:
xialang can you share work, I have tried but its getting error

GeospatialDaryl, 2016-03-07:
Hi-

I've put a fair amount of effort into making the ODM structure platform
independent.

Other than OpenSfM, most of the components compile pretty readily on VS2013
- with some tweaking.  Thinking long-term, there are some basic conventions
in python programming that can ensure cross-comparabilit (e.g. use os.sep,
etc).

Daryl

dakotabenjamin, 2016-03-09:
@ahmadusman can you post your errors? I'd like to get around to this at some point in the near future.

ahmadusman, 2016-03-09:
i have migrate 11 dependencies from Linux to Windows, i manually install those but still keeps complaining about missing dependencies everytime I build.

ahmadusman, 2016-03-09:
Missing Dependencies

xialang2012, 2016-03-17:
@GeospatialDaryl, yes, the problem is the openSFM, but we can used visualSFM or openMVG to replace it. (visualSFM is not an open-source project...)

dakotabenjamin, 2016-03-17:
Given the modular nature of the project, if in the future we want to get Windows support fully and OpenSfM isn't supported by then, we could probably sidle openMVG in as a replacement for windows builds. I think OpenSfM is the best option in general so it shouldn't be replaced.

closed 2016-06-16 by xialang2012