OpenDroneMap/ODM #251: [edgarriba-cmake] ImportError: libecto.so.0.6

Fi156, 2016-02-19:
From gitter:

>cd path/to/odm/dir
>git clone https://github.com/edgarriba/OpenDroneMap.git .
>git checkout cmake
>export PYTHONPATH=$PYTHONPATH:`pwd`/SuperBuild/install/lib/python2.7/dist-packages
>export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:`pwd`/SuperBuild/install/lib
>bash configure.sh
>mkdir build && cd build && cmake .. && make

The line:
>git checkout cmake

Returns:
`git checkout cmake`
`error: pathspec 'cmake' did not match any file(s) known to git.`

So we need to change the line to:
> git checkout edgarriba-cmake

When calling run.py from the git root directory:
(I had to give run.py execution parameters manually.)
`../OpenDroneMap/run.py --project-path ./`                                    
`Traceback (most recent call last):`
`  File "../OpenDroneMap/run.py", line 8, in <module>`
`    import ecto`
`  File "/media/simon/dataSSD250gb/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>`
`    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup`
`ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory`

Is this the way it is supposed to work?

dakotabenjamin, 2016-02-22:
What's the output for `echo $LD_LIBRARY_PATH`?

dakotabenjamin, 2016-02-22:
Check if 6710f44 fixed it

Fi156, 2016-02-23:
On Mint and Ubuntu your fix does not solve the problem.

The `$LD_LIBRARY_PATH` path is empty when starting a new terminal window.
In the terminal where I did the installation in it shows:
`:/media/simon/dataSSD250gb/OpenDroneMap/SuperBuild/install/lib`

When starting with a new terminal window it works as long as I write this path again:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/media/simon/dataSSD250gb/OpenDroneMap/SuperBuild/install/lib
```

Mint 17.2, based on Ubuntu 14.04
Ubuntu 14.04.4 LTS

dakotabenjamin, 2016-02-23:
OK. 

The best thing to do then is write that into your `~/.bashrc` script. Open that file up in your favorite text editor and append 

```
export PYTHONPATH=$PYTHONPATH:/media/simon/dataSSD250gb/OpenDroneMap/SuperBuild/install/lib/python2.7/distpackages
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/media/simon/dataSSD250gb/OpenDroneMap/SuperBuild/install/lib
```

That way every time you open a terminal, those variables will be set.

closed 2016-02-23 by Fi156

Fi156, 2016-02-23:
I had this problem on an freshly installed Ubuntu in a VM.

So I think this is not a specific problem for my system.

dakotabenjamin, 2016-02-23:
Yeah, I'm not exactly sure what is best practice for permanently setting variables from an installation