OpenDroneMap/ODM #253: [edgarriba-cmake] Resizing, Exiftags differ for different cameras

Fi156, 2016-02-23:
```
[DEBUG] Resizing dataset to: /media/simon/dataSSD250gb/Merklingsen1Entzerrt/Jpg90Pics/images_resize
Traceback (most recent call last):
  File "run.py", line 34, in <module>
    plasm.execute(niter=1)
  File "/media/simon/dataSSD250gb/OpenDroneMap/scripts/resize.py", line 68, in process
    new_meta['Exif.Photo.PixelXDimension'].value = img_r.shape[0]
  File "/usr/lib/python2.7/dist-packages/pyexiv2/metadata.py", line 209, in __getitem__
    return getattr(self, '_get_%s_tag' % family)(key)
  File "/usr/lib/python2.7/dist-packages/pyexiv2/metadata.py", line 169, in _get_exif_tag
    _tag = self._image._get
```

The error happens here:
https://github.com/OpenDroneMap/OpenDroneMap/blob/edgarriba-cmake/scripts/resize.py#L68
`Exif.Photo.PixelXDimension` is not a valid Exiftag for Olympus cameras.

I would have suggested the tag `Image Size` instead but this can not be read by pyexiv2.
However exiftool prints this tag for the testdataset as well as for the Olympus data.

Olympus offers the tag `Exif Image Width` and `Exif Image Height` but again this tag is present only when using exiftools, because most tags are named like `Exif.OlympusFi.0x120a` when fetching the tags with pyexiv2.

I see the following solutions:

- Testing gexiv2, because according to http://tilloy.net/dev/pyexiv2/ pyexiv2 its already deprecated.
(And hoping that we can solve the issue that way.)

- Making a database for each manufacture/model for the correct tags

- Thinking of other possibilities excluding exiftags.

dakotabenjamin, 2016-02-23:
Is it only Olympus cameras that don't use Exif standards, or are other cameras going to be a problem too?

dakotabenjamin, 2016-02-23:
I think gexiv2 is probably the way to go in this case.

Fi156, 2016-02-23:
Seems to be an error with my preprocessing pipeline. 
The RAW Olympus images are featuring the necessary tags too!

closed 2016-02-23 by Fi156

dakotabenjamin, 2016-02-23:
Ok! While I still think we should update to modern software, I am glad it is not critical to fix now.