OpenDroneMap/ODM #265: [edgarriba-cmake] Use of LD_LIBRARY_PATH system variable

dakotabenjamin, 2016-02-29:
is [generally not recommended](http://tldp.org/HOWTO/Program-Library-HOWTO/shared-libraries.html) for production. How should we load the libraries properly?

smathermather, 2016-02-29:
Now that we have cmake, do we do a full install on everything? Or, if we are trying to co-exist with other installs, e.g. of OpenCV, etc., what is our best approach?

GeospatialDaryl, 2016-02-29:
It may make sense to leave a default configuration of 'install everything'
for the container folks.  It should be pretty trivial to edit the local
cmake cache to configure for local installs.

Daryl

dakotabenjamin, 2016-02-29:
Do you mind taking a look at it? This is my first project that uses cmake, so I'm not too savvy. 

The relevant files should be here:
https://github.com/OpenDroneMap/OpenDroneMap/blob/edgarriba-cmake/SuperBuild/cmake/External-Ecto.cmake
https://github.com/OpenDroneMap/OpenDroneMap/blob/edgarriba-cmake/SuperBuild/CMakeLists.txt

GeospatialDaryl, 2016-02-29:
Hi Dakota-

I absolutely would/will - but I'm at a training in W VA and have no access
to a linux box.  I'll take a look but not sure I can work it through
without being able to sandbox and test.

I'll follow up - Daryl

dakotabenjamin, 2016-03-02:
If you have ideas/a path forward but no way to test those ideas, feel free to share them with me. With your guidance I may be able to work it out.

GeospatialDaryl, 2016-03-02:
Doug N brought his linuxy laptop - we'll take a peek at this tonight.
Thanks!

Daryl

smathermather, 2016-03-03:
Awesome.

smathermather, 2016-03-03:
(And say hi!)

dakotabenjamin, 2016-03-16:
@GeospatialDaryl any progress?

closed 2020-04-14 by smathermather