OpenDroneMap/ODM #268: Error with processing - vagrant

milamat, 2016-03-07:
What is wrong?

Full matching 1255 / 1285, time: 1.62652206421s
DJI_0396.JPG - DJI_0405.JPG has 740 candidate matches
Robust matching time : 0.000708103179932s
Full matching 706 / 740, time: 1.71885800362s
2016-03-07 16:11:57,050 reading features
2016-03-07 16:11:58,236 Merging features onto tracks
2016-03-07 16:11:59,036 Good tracks: 27864
images 24
nonfisheye images 24
0 partial reconstructions in total.
Traceback (most recent call last):
File "/odm_app/OpenDroneMap/src/OpenSfM/bin/mesh", line 23, in 
reconstructions = data.load_reconstruction()
File "/odm_app/OpenDroneMap/src/OpenSfM/opensfm/dataset.py", line 291, in load_reconstruction
with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: 'opensfm/reconstruction.json'
b

quitting cause:
PYTHONPATH=/odm_app/OpenDroneMap/lib/python2.7/dist-packages "/odm_app/OpenDroneMap/src/OpenSfM/bin/run_all" ope
nsfm
returned with code 256.

vagrant@vagrant-ubuntu-trusty-64:/vagrant_data/odm_data/dronik$

dakotabenjamin, 2016-03-07:
It looks like OpenSfM didn't install properly. Try running the installation script again and post the logs.

milamat, 2016-03-07:
[log.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/161879/log.txt)

dakotabenjamin, 2016-03-07:
Ah, it failed at ceres. The ceres build logs should be found in the logs folder in your odm project directory.

milamat, 2016-03-07:
I am sorry, but what does it mean? What should I do?

dakotabenjamin, 2016-03-07:
there should be a file here: /odm_app/OpenDroneMap/logs/ceres_1_build.log

Please post that so I can see what is failing.

milamat, 2016-03-08:
[log_c.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/163031/log_c.txt)

dakotabenjamin, 2016-03-08:
Is that the full log? I was expecting to see some error.

In any case, this error has popped up before: #198 but I don't know if that solution solved it. My hunch is that there was an installation error, but perhaps there are matching issues as well.

dakotabenjamin, 2016-03-16:
Are you still having this issue, @milamat?

milamat, 2016-03-22:
Hello, I still have the same problem.. I was trying to figure it out where is the problem, but I do not know.. Thank you for trying to help me.

dakotabenjamin, 2016-03-22:
I need more information. Can you try to post both ceres log files? You can copy the files into your /vagrant_data folder and then accesst them on your windows machine from the vodm_data folder you made.

milamat, 2017-01-20:
Thank you a lot! I appreciate it!

closed 2018-03-23 by smathermather