OpenDroneMap/ODM #275: [edgarriba-cmake, 16.04] No module named opensfm.exif

merkato, 2016-03-16:
For some reasons I've installed Xubuntu 16.04 beta. Trying to run OpenDroneMap - some thoughts.

1)  Python-Setuptools not installed, with clean system and ODM ./configure.sh.

This helped: `wget https://bootstrap.pypa.io/ez_setup.py -O - | sudo python`

2) ODM configured, "script finished", but:
```
Traceback (most recent call last):
  File "/home/mechanik/dev/OpenDroneMap/run.py", line 10, in <module>
    from scripts.odm_app import ODMApp
  File "/home/mechanik/dev/OpenDroneMap/scripts/odm_app.py", line 5, in <module>
    from opendm import types
  File "/home/mechanik/dev/OpenDroneMap/opendm/types.py", line 5, in <module>
    from opensfm.exif import sensor_string
ImportError: No module named opensfm.exif
```
edgarriba-cmake branch,```
 git rev-parse HEAD
8b051ebfb589ab14a791d3ac56f86d0ed82e08a2
```

neacao, 2016-03-16:
the same error

dakotabenjamin, 2016-03-16:
Make sure SuperBuild/src/opensfm is in your PYTHONPATH

merkato, 2016-03-16:
Added these two `export` lines to helper script and it's working again. Thank you very much ;)

```
export PYTHONPATH=$PYTHONPATH:pwd/SuperBuild/install/lib/python2.7/dist-packages:/home/mechanik/dev/OpenDroneMap/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/mechanik/dev/OpenDroneMap/SuperBuild/install/lib
/home/mechanik/dev/OpenDroneMap/run.py --odm_georeferencing-useGcp --time --project-path /home/mechanik/dev/odm_data/odm_boruszyn_kap/ --odm_georeferencing-gcpFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/gcp_list.txt
```

dakotabenjamin, 2016-03-16:
I noticed a typo in your code there:
`pwd` should have tick marks around it

merkato, 2016-03-16:
typo corrected....

But...
```
[INFO]    Running ODM PMVS Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /home/mechanik/dev/odm_data/odm_boruszyn_kap/odm_meshing/odm_mesh.ply
[DEBUG]   running /home/mechanik/dev/OpenDroneMap/build/bin/odm_meshing -inputFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/pmvs/recon0/models/option-0000.ply -outputFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/odm_meshing/odm_mesh.ply -logFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9
sh: 1: /home/mechanik/dev/OpenDroneMap/build/bin/odm_meshing: not found
[ERROR]   quitting cause: 
	/home/mechanik/dev/OpenDroneMap/build/bin/odm_meshing -inputFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/pmvs/recon0/models/option-0000.ply -outputFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/odm_meshing/odm_mesh.ply -logFile /home/mechanik/dev/odm_data/odm_boruszyn_kap/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9
returned with code 32512.

An error occurred. Check stdout above or the logs.
```
yes, I know, I should get some coffee and go for coding, but no way. On Friday I have very important exam in my railway job. I sholud start learning on railway rolling stock construction rules ;)

dakotabenjamin, 2016-03-16:
it looks like you haven't built the internal modules. The steps to install ODM right now are:
```
cd path/to/odm/dir
git clone https://github.com/OpenDroneMap/OpenDroneMap.git .
export PYTHONPATH=$PYTHONPATH:`pwd`/SuperBuild/install/lib/python2.7/dist-packages:`pwd`/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:`pwd`/SuperBuild/install/lib
bash configure.sh
mkdir build && cd build && cmake .. && make && cd ..
```

That last line will install odm_x modules

merkato, 2016-03-20:
Ok. I missed this last line ;) Everything is working ;)

closed 2016-03-20 by merkato

rajsaurav1234, 2018-04-24:
.bash

export PYTHONPATH=$PYTHONPATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages
export PYTHONPATH=$PYTHONPATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/install/lib

error -
Traceback (most recent call last):
  File "run.py", line 11, in <module>
    from scripts.odm_app import ODMApp
  File "/home/ubuntu/IIRS/OpenDroneMap/scripts/odm_app.py", line 5, in <module>
    from opendm import types
  File "/home/ubuntu/IIRS/OpenDroneMap/opendm/types.py", line 5, in <module>
    from opensfm.exif import sensor_string
ImportError: No module named opensfm.exif

jamalov, 2018-09-13:
> .bash
> 
> export PYTHONPATH=$PYTHONPATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages
> export PYTHONPATH=$PYTHONPATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/src/opensfm
> export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/ubuntu/IIRS/OpenDroneMap/SuperBuild/install/lib
> 
> error -
> Traceback (most recent call last):
> File "run.py", line 11, in 
> from scripts.odm_app import ODMApp
> File "/home/ubuntu/IIRS/OpenDroneMap/scripts/odm_app.py", line 5, in 
> from opendm import types
> File "/home/ubuntu/IIRS/OpenDroneMap/opendm/types.py", line 5, in 
> from opensfm.exif import sensor_string
> ImportError: No module named opensfm.exif

Did u solve this problem?